react-native-camera/react-native-camera #1689: Android dependency 'com.android.support:exifinterface' has different version for the compile (26.1.0) and runtime (27.1.0) classpath. You should manually set the same version via DependencyResolution

piyushdesuvit, 2018-07-11:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

NielsLindberg, 2018-07-15:
Had similar issue mby the following will help:

In android/build.gradle:
```
ext {
    buildToolsVersion = "27.0.3"
    minSdkVersion = 16
    compileSdkVersion = 26
    targetSdkVersion = 26
    supportLibVersion = "26.1.0"
    googlePlayServices = "15.+"
}

```
In Android/App/build.gradle:
```
dependencies {
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation 'com.android.support:multidex:1.0.0'
    implementation(project(':react-native-firebase')) {
        transitive = false
    }
    implementation "com.google.firebase:firebase-core:${rootProject.ext.googlePlayServices}"
    implementation "com.google.firebase:firebase-auth:${rootProject.ext.googlePlayServices}"
    implementation "com.google.firebase:firebase-firestore:16.0.0"
    implementation(project(':react-native-maps')){
        exclude group: 'com.google.android.gms', module: 'play-services-base'
        exclude group: 'com.google.android.gms', module: 'play-services-maps'
    }
    implementation "com.google.android.gms:play-services-base:${rootProject.ext.googlePlayServices}"
    implementation "com.google.android.gms:play-services-maps:${rootProject.ext.googlePlayServices}"
    implementation project(':react-native-vector-icons')
    implementation "com.android.support:appcompat-v7:${rootProject.ext.supportLibVersion}"
    implementation project(':react-native-linear-gradient')
    implementation (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        implementation 'com.android.support:exifinterface:27.+'
        implementation ("com.google.android.gms:play-services-vision:${rootProject.ext.googlePlayServices}") {
            force = true
    }
}

```

jbonino, 2018-07-17:
hey @NielsLindberg 
You approach worked for me but i have to change one thing.

In android/build.gradle, 

I changed googlePlayServices = "15.+"
to googlePlayServicesVersion = "15.+"

And everything worked correctly

piyushdesuvit, 2018-07-17:
HI @NielsLindberg 
I did all that you mentioned..
 every thing build successfully but apps keeps stopping
[image]

reilem, 2018-07-17:
I had this issue aswell, and what fixed it for me was changing:
`implementation 'com.android.support:exifinterface:25.+'` to `implementation 'com.android.support:exifinterface:27.+'` and adding `supportLibVersion = "26.1.0"` under `ext`.

sibelius, 2018-07-17:
make sure all support libs matches

closed 2018-07-17 by sibelius