react-native-camera/react-native-camera #1694: with react-navigation it's lag

banli17, 2018-07-12:
when navigation.navigate a new screen  ,this screen has RNCamera ，its lag

banli17, 2018-07-13:
@n1ru4l   when toggle page，it is not smooth。
[image]
first page code:
```
navigation.navigate({
      routeName: 'ToolCamera'
})
```

secode page code:
```
renderCamera = () => {
        return (
            <RNCamera
                ref={ref => {
                    this.camera = ref;
                }}
                style={{position: 'absolute', width: WinWidth, height: WinHeight, left: 0, top: 0}}
                permissionDialogTitle={'Permission to use camera'}
                permissionDialogMessage={'We need your permission to use your camera phone'}
            >
            </RNCamera>
        )
    }

    render() {
        let {show} = this.state
        return (
            <View style={{flex: 1, backgroundColor: '#000'}}>
                {this.renderCamera()}
            </View>
        )
    }
```

when i wrap RNCamera with Modal，it's smooth，but cannot back android

gluons, 2018-07-18:
Workaround for me.  
Try to delay renderring camera a few ms.

```jsx
	constructor(props) {
		super(props);
		
		this.state = {
			visible: false
		};
	}

	componentDidMount() {
		setTimeout(() => {
			this.setState({
				visible: true
			});
		}, 300); // Delay 300 ms
	}

	render() {
		return (
			<View>
				{this.state.visible && this.renderCamera()}
			</View>
		);
	}
```

banli17, 2018-07-21:
@gluons   use a Modal wrap it  and place it in current page。it's ok。 my code below。

**Camera.js**
```
import React, {Component} from 'react';
import {
    StyleSheet,
    Modal as RNModal,
    Alert
} from 'react-native';
import {RNCamera} from 'react-native-camera';
import ImagePicker from 'react-native-image-crop-picker'
import {Modal} from 'antd-mobile-rn'
import {toolApi} from 'api'

export default class Camera extends Component {

    static defaultProps = {
        cameraCallback() {
        }
    }

    constructor(props) {
        super(props)

        this.state = {
            modalVisible: false,
        }
        this.key = ''
    }

    async getMenu() {
        let res = await mineApi.IndexMenu()

    }

    async componentWillMount() {

    }

    callback = () => {
        this.props.callback && this.props.callback()
    }

    /**
     * 从相册选择
     */
    selectFromAlbum = () => {
        ImagePicker.openPicker({cropping: false}).then(async (image) => {
            let formData = new FormData()
            let file = {uri: image.path, name: 'hi.png', type: 'multipart/form-data'}
            formData.append('file', file)
            let res = await toolApi.uploadImage(formData)

            this.props.cameraCallback(this.key, image.path)
        })
    }

    /**
     * 拍照
     */
    takePicture = () => {
        this.setState({
            modalVisible: true
        }, () => {
            this.props.cameraCallback()
        })
    }
    /**
     * 选择相机或拍照
     */
    dialog = (key) => {
        this.key = key

        Modal.operation([
            {
                text: '从相册选择',
                onPress: this.selectFromAlbum
            },
            {
                text: '拍照',
                onPress: this.takePicture
            }
        ]);
    }

    cancel = () => {
        this.setState({
            modalVisible: false
        })

    }

    render() {
        return (
            <RNModal
                animationType="fade"
                transparent={false}
                visible={this.state.modalVisible}
                onRequestClose={this.cancel}>
                <RNCamera
                    style={{flex: 1}}
                    ref={ref => {
                        this.camera = ref;
                    }}
                    permissionDialogTitle={'授权使用相机'}
                    permissionDialogMessage={'需要相机权限'}
                >
                </RNCamera>
            </RNModal>
        )
    }
}

const
    styles = StyleSheet.create({
        idcard: {
            height: 200,
            backgroundColor: 'red',
            marginBottom: 10,
            borderRadius: 10
        }
    })
```

## use 

```
<Camera ref={c => this.camera = c} callback={this.submit} cameraCallback={this.cameraCallback}/>
```

SimonErm, 2018-09-25:
Can this be closed?

teggno, 2019-02-08:
I had to display the camera as part of a screen so couldn't use the `<Modal>` workaround. However, I found a nice solution that worked as follows: 

1. In my screen component's constructor:
```js
this.state = {
  cameraVisible: false
};

const subscription = this.props.navigation.addListener("didFocus", () => {
  this.setState({ cameraVisible: true });
  subscription.remove();
});
```

2. My screen component's `render()` function:
```js
render() {
  return (
    <View>
    {/*...*/}
      this.state.cameraVisible ? <RNCamera {....} /> : null
    {/*...*/}
    </View>
  );
}
```

n1ru4l, 2019-02-11:
Going to close this, since this question seems resolved. If you still experience any issues please ask on stackoverflow or open a new issue.

closed 2019-02-11 by n1ru4l