react-native-camera/react-native-camera #1699: Error: ENOENT: no such file or directory, uv_cwd

luvnish19, 2018-07-16:
I am doing setup and executed comamnd : npm install react-native-camera --save. Immediately I get this Error: 

path.js:1167
          cwd = process.cwd();
                        ^
Error: ENOENT: no such file or directory, uv_cwd
    at Object.resolve (path.js:1167:25)
    at Function.Module._resolveLookupPaths (module.js:424:17)
    at Function.Module._resolveFilename (module.js:541:20)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at /usr/local/lib/node_modules/npm/bin/npm-cli.js:19:21
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/bin/npm-cli.js:148:3)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)

jgcmarins, 2018-07-16:
it looks like this is a problem with your npm

jgcmarins, 2018-07-17:
what is the result of `npm -v`?
if it's running ok, you can run again `npm install react-native-camera --save`.

sibelius, 2018-07-17:
this is a problem with npm

try to use `nvm`

closed 2018-07-17 by sibelius