react-native-camera/react-native-camera #1703: Android cannot find symbol new RNCameraPackage()

gegham1, 2018-07-19:
"react-native": "~0.55.2",
"react-native-camera": "^0.6.0",

/android/app/src/main/java/com/bonusapp/MainApplication.java:27: error: cannot find symbol
            new RNCameraPackage()
                ^
  symbol: class RNCameraPackage
2 errors
:app:compileDebugJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

jgcmarins, 2018-07-20:
it looks like you did not link well

gegham1, 2018-07-20:
No I did 
`react-native link react-native-camera                                                          ✓  4423  10:42:00
Scanning folders for symlinks in /Users/hayksafaryan/projects/bonusApp/node_modules (18ms)
rnpm-install info Platform 'ios' module react-native-camera is already linked
rnpm-install info Platform 'android' module react-native-camera is already linked`

jgcmarins, 2018-07-20:
`0.6.0` version is too old, can you try master branch?

gegham1, 2018-07-20:
I tried the master branch and now I have this error

FAILURE: Build failed with an exception.

* Where:
Build file '/Users/hayksafaryan/projects/bonusApp/node_modules/react-native-camera/android/build.gradle' line: 8

* What went wrong:
A problem occurred evaluating project ':react-native-camera'.
> Could not find method google() for arguments [] on repository container.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

tikavane, 2021-03-17:
> https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md

Thank You very much, @sibelius