react-native-camera/react-native-camera #1706: Camera not recognising barcodes on iOS until you rotate the phone

ognjetina, 2018-07-24:
### Which implementation are you using
RNCamera

### Expected behaviour
Scanning barcode using iOS device (portrait mode) should trigger function onBarCodeRead.

### Actual behaviour
App is not able to recognise barcodes until you rotate the phone to the left. (Than it works fine)

### Environment
React native: 0.56.0

### react-native-camera
react-native-camera:  ^1.1.4

sofianegargouri, 2018-07-28:
Same problem, but rotating doesn't fix it.

*Device:* Apple iPad Air 2
*iOS version:* 11.2.1

ognjetina, 2018-07-30:
It seams to be the problem with barcode density...
I do not know if this is the reason but it seams like recognising area is small and if barcode is too dense
you have to move your phone away from barcode so ti would fit in recognising area, and by doing this barcodes that are having hight density become hard to read... (barcode probably looks like one big merged black line...)
This is just a wild guess since it works fine with short 128 barcodes (being short makes them fit in recognising area)

sofianegargouri, 2018-08-02:
I don't know about density. I'm using QR Codes only, and moving the camera closer or further of the code doesn't change a thing

qingjiaowochengzd, 2018-09-22:
+1

iOS  10.2.1
react-native 0.55.4
react-native-camera: 1.2.0

Troubled for a long time，
rotate the ipad4 fixed it !!!

werbeckes, 2018-10-12:
We're also seeing this issue after upgrading react-native-camera and moving from RCTCamera to RNCamera.  However, we are only seeing the issue on android NOT on iOS devices.  Testing with iOS devices scan normally in portrait mode.
react-native: 0.53.0
react-native-camera: 1.3.0
devices: Samsung S6, Motorola Droid Turbo
android versions: 7.0, 6.0.1

@sibelius Is this an issue you've looked into or have suggestions for workarounds?  We were [forced to upgrade](https://developer.android.com/distribute/best-practices/develop/target-sdk) some things to release app updates in November (VERY SOON!).  In that process, RCTCamera broke and we moved to RNCamera.  If it's unlikely this issue will be addressed before that please let me know so we can start exploring other options.  Thanks.

bartello1982, 2018-11-26:
Are there any news about this one? I have the same issue.

paomosca, 2018-12-12:
Today i made some tests (because i noticed a new release [1.6.0], but the problem still persists)

I made an experiment with react-native-orientation-locker, locking the camera in landscape (right or left) make the barcode recognition works again. 
If the app is locked in portrait mode only, the barcode scanning is not working correctly.

I hope this info can help to investigate a little bit more.  

thanks!

n1ru4l, 2018-12-12:
@paomosca What happens if you use the `orientation` option?

paomosca, 2018-12-12:
I'll try to debug a little bit more on the native side... if I'll find any new hint, I'll be back to you... I hope so...

sibelius, 2018-12-14:
try master

sibelius, 2018-12-14:
https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.3

paomosca, 2018-12-17:
@sibelius just tried 1.6.3 but same behavior... nothing changes, is really hard to get barcode scanning in portrait mode (almost impossible)

sibelius, 2018-12-17:
share some logs

paomosca, 2018-12-17:
@sibelius the problem is that i have no log, because the event just doesn't fire. 
(Using iphone 6s with OS 12.1.1)

sibelius, 2019-04-18:
try on latest version

it uses MLKit

aGORyan, 2019-05-03:
Definitely having issues related to barcode density.

Android worked great. iOS I installed via react-native-link rather than CocoaPods. I'm using RNCamera but not MLKit (because I generally dont touch thr Profile).

I can scan simple PDF417, but the ones I'm scanner are denser and the event never fires.

I'm trying to tweak the focus in the hopes that focusing very close to the sense will allow more visibility. If that doesnt work... has anyone confirmed MLKit works better in this scenario?

MateusAndrade, 2019-05-04:
> ones

I tested MLKit on Android and i can confirm that its works better on some cases. Also, there is some problems related to orientation that need to be checked. I going to give a try on some issues to find whats happening. Btw, i recommend using cocoapods to manage your deps on IOS.

aGORyan, 2019-05-08:
Just to confirm... there's no way to get MLKit installed without Cocoapods, right?

I've been at this for a few days now and still having trouble - React/Yoga duplicate ARM64. Seems to be a pretty common problem, but none of the suggestions I'm finding are helping.

My podfile (though I've tried many different variations...):

`
# Uncomment the next line to define a global platform for your project
# platform :ios, '9.0'
# use_frameworks!

target 'DriveAGo' do
# tried without this, just like this, and the more verbose version in the README
 pod 'React', :path => '../node_modules/react-native'
 pod 'Yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

# these were already here
 pod 'Stripe', '~> 10.1.0'
 pod 'AppsFlyerFramework'
 pod 'GoogleMaps'
 pod 'Firebase/Core', '~> 5.0'
 pod 'GoogleTagManager'

# only way to get mlkit is cocoapods
 pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'BarcodeDetectorMLKit'
 ]
 inherit! :search_paths
end

# this seems to fix everyone's problems except mine
post_install do |installer|
 installer.pods_project.targets.each do |target|
   targets_to_ignore = %w(React yoga Yoga)
   if targets_to_ignore.include? target.name
     target.remove_from_project
   end
 end
end
`

sibelius, 2019-05-08:
> Just to confirm... there's no way to get MLKit installed without Cocoapods, right?

I think you can, we have a bash script to download MLKit and linked the project using that

example for Firebase integration

```jsx
if [ ! -f ios/Firebase/Firebase.h ]; then
    mkdir scripts/dirtemp
    mkdir ios/Firebase
    curl -o scripts/dirtemp/temp.tar.gz 'https://dl.google.com/firebase/sdk/ios/5_9_0/Firebase-5.9.0.zip'
    tar -xvzf scripts/dirtemp/temp.tar.gz -C scripts/dirtemp
    cp -r scripts/dirtemp/Firebase/Analytics/* ios/Firebase
    cp -r scripts/dirtemp/Firebase/Messaging/* ios/Firebase
    cp -r scripts/dirtemp/Firebase/Firebase.h ios/Firebase
    rm -rf scripts/dirtemp
fi
```

rickyfabians, 2019-05-10:
any update? 
i'm using the last version, but still got this issue in ios, android works fine.

aGORyan, 2019-05-17:
@rickyfabians 

I'm still stuck. Even tried outsourcing.

I'm considering recreating my 3-year-old project from scratch...the RN update is harsh, though, so I expect we'll need to find other solutions for libraries that break along the way.

Would give my left leg for react-native link to install the MLKit on iOS.

FuriioS, 2019-05-21:
Same problem here.

When I scan the barcodes on an Android device all works perfectly.
But if I use an iOS device, I need to get close my lense to the barcode or that can not be read.

Any sugestion?

Regards,
David

zumaawad, 2019-06-16:
Be sure That Camera view's size is 3:4 ratio (width/height) and it will work well
`      width: deviceWidth,
        height: (deviceWidth*4/3)
`

HelloJack0, 2019-07-13:
@zumaawad 
Thanks for your tips, what is the ratio of iPad? 3:4 seems to be invalid

Amirh24, 2019-11-21:
I still have problem with this. Trying to read a QRcode using IOS and I get nothing. Using 

```
            <RNCamera
                captureAudio={false}
                onGoogleVisionBarcodesDetected={this.barcodeRecognized}
                style={{
                    flex: 1,
                    width: '100%',
                    height: (deviceWidth*4/3)
                }}
            >
            </RNCamera>
```

Where` this.bracodeRecognized` is

```
    barcodeRecognized = (event: { barcodes: Barcode[] }) => {
        event.barcodes.forEach(barcode => {
            console.log(barcode.data);
        })
    };
```

And I get no logs. not even if I am very close or rotating the phone.
Any solution for this? 
I don't want to try MLKIT because to my knowledge you need Cocoapods for this and we user carthage. If there is another way to download it than I can try it.

Thanks!

klosowsk, 2019-11-24:
Having the same issues, works well on android but on iphone I'm facing problems to detect some barcodes.

fabricio0915, 2020-01-15:
Having the same issues, when i rotate my iphone works very well, on android doesn't matter what angle my phone is. Why this topic is closed? anyone already has fixed?

MebDevelop, 2020-02-14:
I am still having issue. Any update?

WrathChaos, 2020-07-06:
Why is this issue is still closed? I'm having this issue on iOS as well. Why nobody take cares of it?

GlenAware, 2020-10-05:
Any update about this issue?

I have the following problem:
On Android: 
onBarCodeRead: not firing
onGoogleVisionBarcodesDetected: working correctly

On iOS:
onBarCodeRead: working correctly
onGoogleVisionBarcodesDetected: not firing

Sidenote: randomly iOS + onGoogleVisionBarcodesDetected picks up a barcode but not reliably.

WrathChaos, 2020-10-05:
> Any update about this issue?
>
> I have the following problem:
> On Android:
> onBarCodeRead: not firing
> onGoogleVisionBarcodesDetected: working correctly
>
> On iOS:
> onBarCodeRead: working correctly [...]

Here is the solution:

https://github.com/react-native-community/react-native-camera/issues/2888#issuecomment-658665220

GlenAware, 2020-10-05:
> > Any update about this issue?
> > I have the following problem:
> > On Android:
> > onBarCodeRead: not firing
> > onGoogleVisionBarcodesDetected: working correctly
> > On iOS:
> > onBarCodeRead: working correctly
> > onGoogleVisionBarcodesDetected: not firing
> > Sidenote: randomly iOS + onGoogleVisionBarcodesDetected picks up a barcode but not reliably.
> 
> Here is the solution:
> 
> [#2888 (comment)](https://github.com/react-native-community/react-native-camera/issues/2888#issuecomment-658665220)

Unfortunately, I already have that in my podfile.

WrathChaos, 2020-10-05:
I've used this method and it works perfect on my side. I've both used this solution on RN 61.5+ and RN 63+. Are you sure you make it right?

GlenAware, 2020-10-06:
> Any update about this issue?
>
> I have the following problem:
> On Android:
> onBarCodeRead: not firing
> onGoogleVisionBarcodesDetected: working correctly
>
> On iOS:
> onBarCodeRead: working correctly [...]

Update: QR Codes work for iOS + onGoogleVisionBarcodesDetected but normal barcodes are not detected.

eshunnar, 2021-01-06:
I'm also having this issue

mengkychen, 2021-01-13:
I have same issue. Any update?

closed 2021-03-21 by stale[bot]

shanu95, 2021-08-01:
Any updates on this?

yorickshan, 2021-12-29:
same issue, any solution?

yorickshan, 2021-12-30:
I'm also having this issue

liho00, 2022-04-03:
same issue +1