react-native-camera/react-native-camera #1707: Permissions problem with React Native v0.56 and Android API<=22

jorgifumi, 2018-07-25:
### Steps to reproduce
1. Create a new react native project with version 56 or higher.
2. Install react-native-camera
3. Open the app in a device with Android 5.1 or lower (API 22) and navigate to camera screen

### Expected behaviour
The preview of the camera should render.

### Actual behaviour
The notAuthorizedView renders.

### Environment
- **React Native version**: 0.56.0
- **React Native platform + platform version**: Android 5.1

### react-native-camera
**Version**: tested in both npm version 1.1.4 and "master" 1.1.5

After getting crazy trying a lot of solutions, I downgraded to React Native 0.55.3 and it started to work as expected. This issue does not happen in iOS or Android API versions 23 and upper.

jgfidelis, 2018-07-25:
It is probably because until RN 0.53 for android <= 23 there was a bug on the PermissionsAndroid lib. It was fixed n RN 0.54 or RN 0.56, someone should send a pr correcting the way we check permissions for android <=23, should not be difficult.

ivosabev, 2018-07-26:
It is fixed by this commit - https://github.com/react-native-community/react-native-camera/commit/07496bb17b641c8b2732615ad40ff6d77b0acb5a

sebastianor, 2018-08-01:
Hello, i still have this problem. My RN version i 0.56 and Android Version i 5.1.1. PermissionsAndroid lib returns that i permissions is granted but i still getting message: "Camera is not authorized".

sibelius, 2018-08-01:
@sebastianor can u try master?

sebastianor, 2018-08-01:
I have installed this by command `npm install react-native-camera --save` so i guess that i've used master branch. I have 1.1.4 version of react-native-camera.

sibelius, 2018-08-01:
neh, master branch has more commits I guess

sebastianor, 2018-08-01:
Could you tell me how to install it ?

n1ru4l, 2018-08-01:
@sebastianor `yarn add react-native-camera@git+https://github.com/react-native-community/react-native-camera.git#860662e0b0e5668503fd6119867abf62b789f677`

sebastianor, 2018-08-01:
It does not help. Camera still not works

sebastianor, 2018-08-01:
Okay it works when you switch on master. Sorry my mistake.

mcclen, 2018-08-02:
I am unable to get RNCamera to work on a Nexus 5 with 4.4.4 installed. The application works really well on emulators above Android 6 and iOS emulators and devices 9.3 and above. The problem seems to be about permissions and the authorization failure occurs for both the Nexus 5 device and any emulator configured at less than Android 6.

Using this to install react-native-camera 1.1.5 -
`
npm install react-native-camera@git+https://github.com/react-native-community/react-native-camera.git#860662e0b0e5668503fd6119867abf62b789f677
`

I have the appropriate permission request per the documentation in the application manifest.

Any help appreciated.

ivosabev, 2018-08-02:
We cannot make it work for anything below API Level 21 (5.0) as well, but it is not the permissions crashing - https://github.com/react-native-community/react-native-camera/issues/1701

afrakhan123, 2018-09-27:
I am having the same problem android 4.4.4 and react native 0.55.4

ortimanu, 2018-11-27:
We are having problems on some Androids OS 5 of our users with this same configuration. Any updates?