react-native-camera/react-native-camera #1708: takePictureAsync leads to a crash on android 21/22

TBouder, 2018-07-25:
### Which implementation are you using
RNCamera

### Steps to reproduce
1. Open a simulator with Lollipop (android 5.1.1 for instance)
2. Open the Camera
3. Take the picture

### Expected behaviour
Should not crash

### Actual behaviour
Crash

### Environment
- **React version**:  16.4.1
- **React Native version**: 0.56.0
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: ^1.1.4 || master

Here is the code we are using. Basicaly it's working perfectly on the newest version of Android/IOS, but on android Lollipop (v21/22) everytime `takePictureAsync` is called, we have a crash (both on simulator and on physical device)
```
takePicture = async () =>
{
	if (!this.camera)
		return;

	const	options = {width: 1080, base64: false, quality: 1};
	const	data = await this.camera.takePictureAsync(options); //CRASH HERE
	//DO STUFF
}

render = () =>
{
	return (
		<RNCamera
			key={'CAMERA'}
			ref={ref => this.camera = ref}
			style={cameraStyle}
			mirrorImage={this.state.cameraType === 'front'} />
	);
}
```

[image]
```
07-25 10:37:58.801 6746-11210/com.shokimap E/unknown:ReactNative: Exception in native call
    java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
        at com.facebook.react.bridge.ReadableNativeArray.getDouble(ReadableNativeArray.java:114)
        at com.facebook.react.bridge.JavaMethodWrapper$4.extractArgument(JavaMethodWrapper.java:64)
        at com.facebook.react.bridge.JavaMethodWrapper$4.extractArgument(JavaMethodWrapper.java:60)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:359)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:160)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
        at android.os.Looper.loop(Looper.java:135)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
        at java.lang.Thread.run(Thread.java:818)
```

mcabs3, 2018-07-25:
I am also having this issue when running on a physical device with Android 7.0.0, however I don't get the error, it just silently crashes.

PetengDedet, 2018-07-27:
Any solution please?
i got this error on physical device running android Lollipop 5.0.2.

faizalshap, 2018-07-27:
Had the same problem. Resolved by following the exact installation step. 
So seems like installation issue.

xenu256, 2018-09-21:
Same with 27. Also, may be related, project builds only without vision api.

SimonErm, 2018-09-22:
I think this could be an issue of react-native, because there is no class directly related to react-native-camera in stacktrace

xenu256, 2018-10-13:
This is related to permissions, see my answer: [https://github.com/facebook/react-native/issues/17985?fbclid=IwAR3sETcagrGvb0OHhgX9KhJnygrM86A9pc8bzqorB_XgUT5tYUvSMWsrQVg#issuecomment-429541452](#17985)

AND also there is an error in an example,. Here's fully working example on RN 0.57 Android 27 solving both of those problems:

```
class App extends Component {
  _requestPermissions = async () => {
    if (Platform.OS === 'android') { 
      const result = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA)
      return result === PermissionsAndroid.RESULTS.GRANTED || result === true
    }
    return true
  }

  _takePicture = async () => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true }
      const data = await this.camera.takePictureAsync(options)
      console.log(data.uri)
    }
  }

  componentDidMount = () => {
    ({ _, status }) => {
      if (status !== 'PERMISSION_GRANTED') {
        this._requestPermissions()
      }
    }
  }

  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={(ref) => { this.camera = ref }}
          style={styles.preview}
          flashMode={RNCamera.Constants.FlashMode.on}
          type={RNCamera.Constants.Type.back} />
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity
            onPress={() => this._takePicture()}
            style = {styles.capture}>
            <Text style={{ fontSize: 14 }}> SNAP </Text>
        </TouchableOpacity>
        </View>
      </View>
    )
  }
}
```

Airomad, 2018-10-24:
I had the same issue. It was because I used **resumePreview** and **pausePreview** methods in my code (which works on iOS without errors)

darekg11, 2018-12-17:
I am on:  
react-native: **0.57.4**
react-native-camera: **1.1.12** because of #1754 

Currently in following scenario:  
1. Open page where `react-native-camera` is mounted for the first time  
2. Decline permissions
3. Click on the button to ask for permissions
4. Accept them
5. Start recording via `await camera.recordAsync(OPTIONS)`
6. Get a crash of message and stack trace like in OP

When I restart the app I can record without any problems.
When I close the page containing the `react-native-camera` and come back it also works without problems.
This is only happening when I go through steps 1-6 and remain on the same page when doing step 5.

Anyone could help? Right now I made a workaround to close the page forcing user to open it once more when user declines the permissions but this is not really user-friendly.

n1ru4l, 2018-12-18:
@darekg11 Could you provide a minimal reproduction example? The best way would be by forking this repository and adjusting https://github.com/react-native-community/react-native-camera/tree/master/examples/basic

Thank you for investigating on this issue.

darekg11, 2018-12-18:
@n1ru4l I would love too but, I am afraid that I won't get a chance until weekend.

chmartinez, 2018-12-20:
In my case, the problem was related to how the `RNCamera` works with FaCC (https://github.com/react-native-community/react-native-camera/blob/master/src/Camera.js#L285).

*NOTES*: 
* this issue only happened to me on physical Android devices with react-native v0.54 and RNCamera v1.3.1
* the code I'm sharing is not fully functional but a way to show what was my scenario and how I fixed the issue.

My scenario was like this:
* Used the `ref` approach (due to some aspect ratio issues I found with the FaCC approach).
** I had a separate button to take the snapshots/photo (outside the `RNCamera`).
* I needed to know what's the status granted so I can show different dialogs to the user and for that, I'm handling the permission request outside the `RNCamera`.

```
// @flow
import React from 'react';
import { Text, TouchableOpacity, View } from 'react-native';
import { RNCamera } from 'react-native-camera';

class MyCamera extends React.Component {
    camera: ?Object = null;

    state = {
        photo: {}
    }

    async componentDidMount() {
        // a callback prop that I use to get permission. It returns a `grant` string with the result of the permission request
        await this.props.requestPermissions('camera');
    }

    takePicture = () => async () => {
        if (this.camera) {
            const photo = await this.camera.takePictureAsync(CAMERA_OPTIONS);
            this.setState({ photo });
        }
    }

    render() {
        if (this.state.photo !== {}) {
            <Preview photo={this.state.photo} />
        }

        return (
            <View style={{ flex: 1 }}>
                <RNCamera ref={(ref) => this.camera = ref} />
                <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                    <TouchableOpacity
                        onPress={() => this.takePicture()}
                    >
                        <Text>Take photo</Text>
                    </TouchableOpacity>
                </View>
            </View>
        );
    }
}
```

* When the `RNCamera` was mounted, its inner state prop `isAuthorized` was always false, so the `props.pendingAuthorizationView` element was getting rendered.
* When the camera permission was granted, my main component wasn't triggering an `RNCamera` re-render. The inner component was the `pendingAuthorizationView`, not the camera itself.
* If you clicked the `Take photo` text, you get the 'virtual method double' error.

*Solution*:  
* store the `grant` value from my requestPermission prop within my component inner state
* conditional rendering for the `RNCamera` (based on the new state prop):

```
// @flow
import React from 'react';
import { Text, TouchableOpacity, View } from 'react-native';
import { RNCamera } from 'react-native-camera';

class MyCamera extends React.Component {
    camera: ?Object = null;

    state = {
        isCameraReady: false,
        photo: {},
    }

    async componentDidMount() {
        const grant = await this.props.requestPermissions('camera');
        if (grant === 'authorized') {
            this.setState({
                isCameraReady: true,
            });
        }
    }

    takePicture = () => async () => {
        if (this.camera) {
            const photo = await this.camera.takePictureAsync(CAMERA_OPTIONS);
            this.setState({ photo });
        }
    }

    render() {
        if (this.state.photo !== {}) {
            <Preview photo={this.state.photo} />
        }

        return (
            <View style={{ flex: 1 }}>
                {this.state.isCameraReady ? <RNCamera ref={(ref) => this.camera = ref} /> : <View><Text>Camera is not ready</Text></View>}
                <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                    <TouchableOpacity
                        onPress={() => this.takePicture()}
                    >
                        <Text>Take photo</Text>
                    </TouchableOpacity>
                </View>
            </View>
        );
    }
}

```

**Why does it work with FaCC?**
Don't know. My guess is that it's a race condition. Maybe the componentDidMount from `RNCamera`  gets executed first or something. I can't tell.

** Why does it only happened on physical devices?**
Guess for the same reason. Race conditions and not having a solid structure around the permissions and the state changes.

Hope it helps someone. Thanks!

darekg11, 2018-12-22:
@n1ru4l 
Forked repo with easy reproduction:
https://github.com/darekg11/react-native-camera/tree/reproduce-the-crash

Reproduction steps:
1. Install the app (notice the permissions are not listed, as any new app should not ask for permissions upon startup when targeting devices with Android 23+)
2. Launch app
3. Decline permissions.
4. Click `REC` button
5. Crash

# What have I changed:
1. Bumped the version in `AndroidManifest.xml` and `build.gradle` to use sdk in version `23` where Permissions are not shown by default when launching it on device with Android 6.0+:  
https://developer.android.com/distribute/best-practices/develop/target-sdk#prelollipop  
2. Moved the rec button to always be visible in order to be able to click `Rec` even when the Camera component fails to initialize

# Logcat stack trace:
```
12-22 13:55:58.770 27995-31188/com.rncameraexample V/AudioManager: playSoundEffect   effectType: 0
    querySoundEffectsEnabled...
12-22 13:55:58.776 27995-31188/com.rncameraexample E/unknown:ReactNative: Exception in native call
    java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
        at com.facebook.react.bridge.ReadableNativeArray.getDouble(ReadableNativeArray.java:114)
        at com.facebook.react.bridge.JavaMethodWrapper$4.extractArgument(JavaMethodWrapper.java:64)
        at com.facebook.react.bridge.JavaMethodWrapper$4.extractArgument(JavaMethodWrapper.java:60)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:359)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:160)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
        at android.os.Looper.loop(Looper.java:150)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
        at java.lang.Thread.run(Thread.java:833)
```
Unfortunately this is everything that there is in log when this error occurs :(
I have attached the debugger in Android Studio and my results:
It must be somewhere n in React-Native bridge because my breakpoints in `react-native-camera` Android code are not reached when that error happens.
As soon as I enable permissions, breakpoints work correctly.

# My system and device info:
```
React Native Environment Info:
    System:
      OS: macOS High Sierra 10.13.6
      CPU: (4) x64 Intel(R) Core(TM) i7-7660U CPU @ 2.50GHz
      Memory: 1.52 GB / 16.00 GB
      Shell: 5.3 - /bin/zsh
    Binaries:
      Node: 8.11.3 - /usr/local/bin/node
      npm: 5.6.0 - /usr/local/bin/npm
    SDKs:
      iOS SDK:
        Platforms: iOS 12.1, macOS 10.14, tvOS 12.1, watchOS 5.1
      Android SDK:
        API Levels: 22, 23, 25, 26, 27, 28
        Build Tools: 23.0.1, 25.0.2, 25.0.3, 26.0.1, 26.0.2, 26.0.3, 27.0.3, 28.0.1
        System Images: android-22 | Google APIs Intel x86 Atom, android-25 | Google APIs Intel x86 Atom, android-26 | Google APIs Intel x86 Atom, android-27 | Google APIs Intel x86 Atom, android-28 | Google APIs Intel x86 Atom
    IDEs:
      Android Studio: 3.1 AI-173.4819257
      Xcode: 10.1/10B61 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.5.0 => 16.5.0
      react-native: 0.57.1 => 0.57.1
    npmGlobalPackages:
      react-native-git-upgrade: 0.2.7
      react-native-svg: 7.1.2
```
Device that is crashing:  
`Huawei P8 Lite, running Android 6.0`

# How I have solved it in project?
I have used FaCC and conditional rendering based on `status` coming from Camera - solution similar to that posted by @chmartinez

pitchourou, 2019-02-28:
not resolved, still crashing with the latest release : 1.12.0

KryDos, 2019-03-07:
Fixed it on my end by removing `skipProcessing: true` from options object of `takePictureAsync`.

pitchourou, 2019-03-07:
> Fixed it on my end by removing `skipProcessing: true` from options object of `takePictureAsync`.

I ended up doing that too, then processing the image using `react-native-image-resizer`

closed 2019-06-07 by stale[bot]