react-native-camera/react-native-camera #1709: Wireless Camera Video Feed

donei003, 2018-07-25:
Hello, first off thank you for your time in reading this. I am building an app that would access a camera's video feed which can be viewed through the app and take stills. After looking for quite sometime I have not been able to find any documentation on this.

Is this feasible with RN? If so, where should I focus my efforts?

Thank you!

n1ru4l, 2018-07-26:
@donei003 First of all thank you for presenting your use case! 

I am pretty sure this is not implemented yet (on either android or ios). Since there are apps available and a quick stackoverflow search gave me at least one example (https://stackoverflow.com/questions/12242513/how-to-get-real-time-video-stream-from-iphone-camera-and-send-it-to-server) this can certainly be implemented.

IMHO out of performance reasons (you would not want to pass every frame of the video over to the js thread to send it to your server/app/whatever) this should be settled on the native thread.

Maybe I am going to far with this statement, but I think this would be a use-case where you would want to go with your own implementation/fork instead of trying to implement it in react-native-camera.

@jgfidelis Anything to add? What is your opinion?

n1ru4l, 2018-07-26:
@donei003 checkout https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543

donei003, 2018-07-27:
@n1ru4l  Thank you for your quick response! 

I think I left out an important part of the project. We are using an endoscope which the phone connects to, to gather its video/image feed from. There are apps on both the Play/App Store called "WiFi View" which achieve this functionality, however, I have not been able to figure out how they have managed this. I reached out to the company that manufacturers the endoscope, Depstech, and they have been unwilling to provide any assistance. 

In my research I have come across https://github.com/oney/react-native-webrtc but that may not be the appropriate solution.

closed 2019-06-08 by stale[bot]

Rajakalym, 2021-05-01:
Hello @donei003. I'm also working on a similar project using an endoscope to which the phone connects to. Using RN for Android/iOS.
Did you manage to achieve this thing using RN? The android apps you mentioned like Wifi look and Wifi view, did this using the WebRTC library I have the android source code of their Wifi Look App.

Andrew-Pynch, 2023-01-11:
> Hello @donei003. I'm also working on a similar project using an endoscope to which the phone connects. Using RN for Android/iOS. Did you manage to achieve this thing using RN? The android apps you mentioned like Wifi look and Wifi view, did this using the WebRTC library I have the android source code of their Wifi Look App.
> 
> Any direction to achieve this using React Native, assistance, or advice would be greatly appreciated! Thanks in advance!

Hey Ghost, I am also working on something similar. Did you have any success with your project.

coucoseth, 2023-04-28:
anyone got a solution to this yet?