react-native-camera/react-native-camera #1712: Barcode not being scanned after first scan

Sterlingryan, 2018-07-27:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. Open camera screen
2. Point to qr code
3. Barcode is scanned and onBarCodeRead is triggered
4. Navigate back to camera screen
5. Point to qr code
6. Nothing is triggered, even after closing and opening the app

### Expected behaviour
1. Open camera screen
2. Point to qr code
3. Barcode is scanned and onBarCodeRead is triggered
4. Navigate back to camera screen
5. Point to qr code
6. Barcode is scanned and onBarCodeRead is triggered again

### Actual behaviour
onBarCodeRead is not being triggered after the first scan, I can only make it work if I uninstall and install the app again on the device. Then the onBarCodeRead is triggered only once. I'm thinking it something related to permissions or library not closing the camera.

### Environment
- **React Native version**: react-native: 0.55.4
- **React Native platform + platform version**: Any Android device

### react-native-camera
**Version**: 1.1.4

priezz, 2018-08-12:
I have the same issue. In my case it is not necessary to reinstall the app (though it helps), but phone reboot also allows to scan the code at the first camera use. App forceful restart does not help.

n1ru4l, 2018-12-19:
@priezz @cameronbell Are you interested in investigating and fixing this issue?

Sterlingryan, 2018-12-19:
@priezz @cameronbell  @n1ru4l  I apologize for the late reply, if it is of any help, I upgraded the version to 1.2.0 (using RNCamera) and it fixed that issue

n1ru4l, 2018-12-19:
@cameronbell Please open a new issue if you are experiencing this issues on the newest version.

closed 2018-12-19 by n1ru4l