react-native-camera/react-native-camera #1713: [Android] there is no possibility to control video quality

djleonskennedy, 2018-07-28:
### Which implementation are you using
*RNCamera*

### Steps to reproduce
1. create app with camera
2. set h264 for quality
3. record video

### Does it work with Expo Camera?
no expo used

### Expected behaviour
Have ability to contoll video quality output on android

### Actual behaviour
No any chance to control video quality output on android

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android 6.0+

### react-native-camera
**Version**: ^1.1.4

### motivation
Our project really need small video size, cause needs to upload videos throw slow internet

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
will do if quality controll will be added to android ;)

n1ru4l, 2018-07-30:
we are open for contributions 😊

djleonskennedy, 2018-07-31:
@n1ru4l I'll do, could provide information about contribution guide? Or how to set up env to test changes library ?

n1ru4l, 2018-07-31:
It is hard to test code that uses real hardware. I would say you start by forking this git repository and start working on a new branch. Once the feature is ready you can create a Pull Request which some maintainers/contributors or users can check out and confirm whether stuff is working as expected or not.

pribeh, 2018-09-13:
@djleonskennedy did you take a stab at this?

djleonskennedy, 2018-09-13:
@pribeh yes, i've tested bitrate, its works different depending on device, but it works, i'll prepare pull request on this holidays

argenisleon, 2018-10-11:
Hi @djleonskennedy, any news on this?. We are open to collaborating to make this happen.

djleonskennedy, 2018-10-17:
@abrahambotros module already have quality profiles, however i'm investigating why they cannot be set.

As i understood this method have to apply profile
[Camera1.java#L826](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L826)

but it doesn't might something undocumented, also we should able to add ability to set custom profile.

Sorry, don't have time for now, will see on it later

djleonskennedy, 2019-06-06:
Closed due to no time to make changes

closed 2019-06-06 by djleonskennedy