react-native-camera/react-native-camera #1714: [Android] text recognition fails on Portrait mode

diogenesgg, 2018-07-30:
### Which implementation are you using

RNCamera 1.1.4

### Steps to reproduce
Text Recognition is not working when in Portrait Mode. I tested with a blank page with the following two lines: 
**Hello World**
**Hello World Again**
And it only detects Single or double characters on each event, like 'O', 'IC'. 

When I rotate to landscape mode, it works perfectly. 

Heres is how I handle the event and render it:
`onTextRecognized = ({textBlocks}) => this.setState({ detectedTexts: textBlocks.map(b => b.value) })`
```
renderDetectedText() {
    return (
      <View style={[styles.facesContainer,{left: 10, top:"50%"}]}>
        <Text style={styles.flipText}>{this.state.detectedTexts.join("\n")}</Text>
      </View>
    )
  }
```
The RNCamera element takes the full screen.

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Installed on a physical Android 7.0 Device

diogenesgg, 2018-08-02:
I was able to solve the problem by modifying some Java code.
I noticed it had to do with the Rotation. It was trying to read the text bottom-up. 

When I would rotate the phone from portrait to landscape, it would still try to recognise from the phone's bottom-up, which is now left to right.

On `org.reactnative.camera.RNCameraView`, (located on node_modules/react-native-camera/android/src ) I changed the `onFramePreview` method in order to remove the following code (commented out):
```
@Override
      public void onFramePreview(CameraView cameraView, byte[] data, int width, int height, int rotation) {
        //int correctRotation = RNCameraViewHelper.getCorrectCameraRotation(rotation, getFacing());
        int correctWidth = width;
        int correctHeight = height;
        byte[] correctData = data;
        //if (correctRotation == 90) {
        //  correctWidth = height;
        //  correctHeight = width;
        //  correctData = rotateImage(data, correctHeight, correctWidth);
        //}
//...
```

I wonder if the rotation is suitable only for some device models.

matt-oakes, 2019-01-10:
I'm seeing this as well on a Nexus 5X. It looks like it's trying to rotate the image before text detection, however, it's not doing it correctly.

When in portrait, the rotation value is set to `90`, however, it only correctly detects text if I lock the screen rotation and then rotate the phone 90 degrees anti-clockwise.

I'm going to look into this a little more.

matt-oakes, 2019-01-11:
I believe I have found the issue.

### The issue
[This code](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L278-L284) in the library is taking in the `rotation` from the `CameraView` and converting it to the image rotation. It is, however, making some assumptions about the camera orientation which is why it is failing on some devices.

The `rotation` from the `CameraView` callback corresponds to the display orientation. Portrait is *usually* `0`, however, that's device dependent. The camera on the device also has an orientation relative to the display orientation. You can read about it [here](https://www.captechconsulting.com/blogs/android-camera-orientation-made-simple).

### The fix
* First, I need to update the [Google CameraView](https://github.com/google/cameraview) library to allow us to get the orientation of the camera for both the deprecated camera API ([here](https://developer.android.com/reference/android/hardware/Camera.CameraInfo.html#orientation)) and the new one ([here](https://developer.android.com/reference/android/hardware/camera2/CameraCharacteristics.html#SENSOR_ORIENTATION)).
* Next, we need to update this library to take the camera orientation into account.

I'll keep you updated with my progress.

n1ru4l, 2019-01-18:
:tada: This issue has been resolved in version 1.8.1 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.8.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2019-01-18 by n1ru4l

carlobambo, 2019-06-14:
I'm using v2.9.0 but I'm still having this issue. I tested using nexus 5x.

PR #2050 (2019-01-18): fix(android): use the correct camera orientation to fix image detection
### The issue
[This code](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L278-L284) in the library is taking in the `rotation` from the `CameraView` and converting it to the image rotation. It is, however, making some assumptions about the camera orientation which is why it is failing on some devices.

The `rotation` from the `CameraView` callback corresponds to the display orientation. Portrait is *usually* `0`, however, that's [... cut]
commit f7fb3e36 (2019-01-18): fix(android): use the correct camera orientation to fix image detection (#2050), Fixes #1714