react-native-camera/react-native-camera #1719: Getting a standard 4:3 aspect ratio?

hampelm, 2018-07-31:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce

On iOS, I'm taking a photo with `takePictureAsync`: 

```
      const data = await this.camera.takePictureAsync({
        exif:               true,
        metadata:           { location, },
        quality:            0.75,
        skipProcessing:     true,
        forceUpOrientation: true,
      })
```

The photo that comes through is at the aspect ratio of the screen. Is there a way to get photos in the standard 4:3 aspect ratio that the native iOS camera app uses? 

### Environment
- **React Native version**: `0.55.4`
- **React Native platform + platform version**: `iOS 11.4`
### react-native-camera
**Version**: `1.1.4`

MD3XTER, 2018-08-15:
The "4:3" quality ratio is only available to Android not iOS.

AndrewHenderson, 2018-10-01:
I'm using a full screen camera view — I find that iOS captures an image in the ratio of the screen. However, Android captures a 16:9 image.

I want the image produced to match the screen ratio on both platforms.

The result on Android is an image with info on the left and right that was not present on the screen when capturing. This is jarring to the user.

Udbhav12, 2018-10-12:
I am facing this exact same problem on Android. Did anyone manage to get it working ?

TheBlackHacker, 2018-12-11:
For Android devices, you shoud set the ratio manualy, see:
https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#user-content-android-ratio

closed 2019-06-08 by stale[bot]