react-native-camera/react-native-camera #1722: User not asked for permissions to use camera android

kanekotic, 2018-08-02:
### Which implementation are you using
 - RNCamera

### Steps to reproduce
1. follow installation and linking for android
2. Use FaCC (Function as Child Components) tutorial for RNCamera
3. Run in android simulator (also tested in phone: one plus 6 )

### Does it work with Expo Camera?
Not using expo. Ejected app.

### Expected behaviour
I expect to have a pop up asking permissions to use the camera.

### Actual behaviour
No pop up. Just stays in status not ready.

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android API 28 Simulator (development in osx 10.13.6)

### react-native-camera
**Version**: 1.1.4

kanekotic, 2018-08-02:
the related code can be found in: https://github.com/cleanlyer/cleanly/blob/master/src/components/report.js#L43-L59

jgcmarins, 2018-08-02:
This is a known bug at react native on android platform. A workaround is to ask permission on app startup.

kanekotic, 2018-08-03:
thanks for the fast answer @jgcmarins. 
What app startup are we talking about react-native (index in js code) or Android (activity in java code)? 
Is there any reference where I can get more information on this bug or the workaround?

kanekotic, 2018-08-03:
for completeness if anyone gets also here:
- Link to explanation of bug: https://stackoverflow.com/questions/45545661/react-native-android-permission-denied
-workaround: for now use the master branch `yarn add "git+https://git@github.com/react-native-community/react-native-camera"`

n1ru4l, 2018-08-03:
Could this be fixed in https://github.com/react-native-community/react-native-camera/commit/ae35e4f2169a01f481ada98a00ceeccc94aa6e56 ?

kanekotic, 2018-08-03:
@n1ru4l I don't think so, I already checked with the master branch that solves the issue of `Granted`. This is related to not even get the user prompt for the permissions.

devapro, 2018-08-03:
I used the master branch. After the first open screen with  RNCamera get the user prompt for the permissions (camera and microphone).  After the user granted permission camera work normally.

jgcmarins, 2018-08-03:
try to ask permission on app startup

kanekotic, 2018-08-05:
thanks, @devapro & @jgcmarins, your answers where insightful.
I did test the master branch before, but as explained this is a different issue that was being solved, at least in my case it did not bring any effect.
On the other side to a manual permission at `startup` did `works`. In any case this but open for me the question what is the difference with this and doing it at the construction of the component. Doing this during the `construction of component` also `works`. I then moved it to `status management` of `RNCamera` and it also worked.
This for me is a symptom there is a bug somewhere in the code of RNCamera.
My current solution until I have time to review the code of `RNCamera` is:

1. helper function:

```js
import { PermissionsAndroid, Platform } from 'react-native'

export const requestPermissions = async () => {
    if(Platform.OS === 'android') { 
        const result = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA)
        return result === PermissionsAndroid.RESULTS.GRANTED || result === true
    }
    return true;
}
```
2. On `RNCamera`:
```js
<RNCamera
    style={styles.preview}
    type={RNCamera.Constants.Type.back}>
                    {({ camera, status }) => {
                        if(status === 'PENDING_AUTHORIZATION'){
                            requestPermissions()
                                .then(result => {
                                    if(!result)
                                        Actions.pop()
                                })
                        }
                    }}
                    </RNCamera>
```
personally, I prefer this than startup because Android seems to be going towards the idea of asking for the permission as per the moment a function is required and not at the beginning.

jgcmarins, 2018-08-05:
Great to know @kanekotic
Any idea on how to fix it on `RNCamera`, just open a pull request \o/

kanekotic, 2018-08-06:
so I have debugged a bit and it has to do with the rationale passed to `PermissionsAndroid.request`, if present it actually does not show.

jgcmarins, 2018-08-06:
https://github.com/facebook/react-native/issues/17985

closed 2019-06-08 by stale[bot]