react-native-camera/react-native-camera #1724: Android recordAsync() crashes

vamvoukas, 2018-08-02:
we use RCTCamera and version 1.1.5 @latest version from git.
In some android devices when we start to record a video with recordAsync(), app crashes.
And we receive this message:
{"framesToPop":1,"code":"E_CAPTURE_FAILED","line":17711,"column":29,"sourceURL":"http://10.0.2.2:8081/index.android.bundle?platform=android&dev=true&hot=false&minify=false"}

### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgcmarins, 2018-08-02:
RCTCamera is deprecated, use RNCamera instead

devapro, 2018-08-10:
maybe this related to permissions. Try manual set permissions for camera and microphone to app. after this try call recordAsync

closed 2018-12-11 by n1ru4l