react-native-camera/react-native-camera #1725: RNCamera Android Orientation

Tsiniloiv, 2018-08-02:
So, RNCamera on Android will frequently save pictures in the wrong orientation. This seems to happen especially when the app's orientation has been locked - in my case, my app is locked to landscape orientation, and pictures will be saved correctly in one orientation, and upside down in the other orientation.

I'm very much not an expert on this, but I've been digging through the code and other posted issues, and near as I can tell this happens because RNCamera is finding its orientation by listening for onOrientationChanged events. (I don't know if locking the orientation blocks these events from firing or what, but like I said, very much not an expert)

The fixOrientation option seems to take the final image, read its real exif orientation, then processes the image to match that, I think? The big problem is that it's slow as hell, especially compared to RCTCamera.

So I figured I'd check in case I've missed a solution or workaround somewhere - has anybody figured out how to fix this, or come up with a workaround that doesn't multiply the amount of time it takes to take a picture?

If not, I _think_ I may have found where the camera orientation change happens. Please correct me if I'm wrong: The RNCameraView implements Google's CameraView, and just straight up calls super.takePicture(). CameraView seems to be a very nice and neat and convenient wrapper around the Camera API, so that's nice, except it bases its orientation on listening for onOrientationChanged events.

So, in theory, adding a check for the device's real orientation to CameraView's takePicture() function might solve the orientation issue without needing to reprocess every single picture.

Heading to lunch now, but I'm gonna try that after. Figured I'd put this up. Am I on totally the wrong path here? Has anybody solved this, so I don't have to waste time being wrong or duplicating someone else's solution?

SimonErm, 2018-08-02:
https://github.com/react-native-community/react-native-camera/issues/1662 i think it's the same like on ios. We should use accelerometer or gyroscope to get orientation. I think the native Android camera does the same.

Tsiniloiv, 2018-08-02:
I don't know anything about iOS coding, unfortunately, but it's a relief to know my train of though isn't terribly wrong.

I did some digging on the Android side and turns out I was mostly wrong, slightly right. CameraView uses yet another wrapper around the Camera API, and that's where the API is really called. The orientation listener thing is solid - it's to rotate the display when the display is supposed to rotate, I think.

Looking into moving the old RCTCamera's adjustCameraRotationToDeviceOrientation() function into Camera1.java now. That's the deepest layer I can find, with access to the Camera instance directly, it looks like. Fingers crossed!

Tsiniloiv, 2018-08-02:
As usual, it turns out things are way more complicated than I thought. (I am very much not an expert)

While trying to do the thing I discovered that Camera1 is actually quite comprehensive and solid, and was already doing more or less everything I thought I needed to make it do.

Gonna try and instead see what I can do with RCTSensorOrientationChecker, which seems like it does what we're all looking for - directly listen to the accelerometer to track the device's movements. Then I'll have to find some way to pass that down to the camera parameters.

If that doesn't work, then I have absolutely no idea what will. There really is no reason the orientation shouldn't be properly being tracked and updated. But even when I try directly setting the camera parameters to some value (and applying the parameters, of course), the final image rotation doesn't change! What!!!

vvusts, 2018-11-21:
Is there any status on this?

n1ru4l, 2018-11-26:
@Tsiniloiv @vvusts You can help fixing this issue by testing the following pull request: https://github.com/react-native-community/react-native-camera/pull/1941

(it is android only)

Tsiniloiv, 2018-11-26:
@vvusts No status on my end. Gave up on this a while back and just locked the app in the orientation that returns the right-side-up images.

@n1ru4l Thanks for the tip! I'll see if I find some time to look at it.

Fsarmento, 2018-11-26:
Hi, I am also having the same problem. With orientation locked to portrait, if the user takes the picture in landscape, I would like the output foto to be rotate in order to be correctly visualized in portrait. That doesn't happens at the moment.

I gave a look at the code before having a look at your pull request and my first approach was to update [calcCameraRotation](https://github.com/react-native-community/react-native-camera/blob/71b10c1ac84b5852728d12412e6ff035a1a692f3/android/src/main/java/com/google/android/cameraview/Camera1.java#L658) method to use real device orientation instead of display orientation.

[DisplayOrientationDetector](https://github.com/react-native-community/react-native-camera/blob/0b816adbfda6edb5d6db088e35a0689ee2175297/android/src/main/java/com/google/android/cameraview/DisplayOrientationDetector.java#L47) callback have the (correct) device orientation. However, inside the callback, `orientation` is ignored and the orientation is obtained with `final int rotation = mDisplay.getRotation();` (getRotation() does not change if screen is locked). 

So, to correct this issue, I added a flag `useDeviceOrientation` option to `takePictureAsync`, that defines which orientation should be used (display or device).

I implemented this changes in a fork [here](https://github.com/Fsarmento/react-native-camera).

Then I discovered your pull request :) My question is, in android, `takePictureAsync` takes an eternity to resolve. I see your solution is to rotate the bitmap image in `ResolveTakenPictureAsyncTask`.

My solution changes the camera parameters of [`Camera mCamera`](https://github.com/Fsarmento/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L391). I did not dig deeper how changing the parameter efects the output of takePicture. However, i did not experienced any extra delays resolving takepictureasync.

SimonErm, 2018-11-26:
i think we should use the deviceOrientation by default and add a orientation option to takePicture, similar to iOS. I already worked on it and will do a PR this week.

n1ru4l, 2018-12-11:
@SimonErm What is your progress on this PR?

closed 2018-12-13 by sibelius

n1ru4l, 2018-12-13:
:tada: This issue has been resolved in version 1.6.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #1992 (2018-12-13): fix(android): use device orientation for images/videos instead of display orientation
This is my attempt for fixing #1725 based on the work of @Fsarmento (https://github.com/react-native-community/react-native-camera/issues/1725#issuecomment-441599916 https://github.com/react-native-community/react-native-camera/compare/master...Fsarmento:master)

This PR unifies the orientation used for the pictures/video with the `ios` implementation by using the device orientation instead of the display orientation.

**Related:**

Fixes #1725
Closes #1941
Fixes #1053

cc @sibelius @JulienUsson @SimonErm @Ts [... cut]