react-native-camera/react-native-camera #1727: Android app crash when using hardware back button.

shahchaitanya, 2018-08-02:
I'm using react-navigation. My Camera Screen is two level deep. When i use back button twice from Camera Screen, app exit and crash.

I'm using RNCamera. My react-native-camera version is 1.1.0, react-native version is 0.53.3.

Expected Behaviour - Successfully Exist .

Actually Behaviour - Exit and Crash. #878

shahchaitanya, 2018-08-02:
@sibelius Thanks for reply back. 
Off Course, It's happening in android. I'm not getting stack trace of Js. Because it's not JS Error. I'm using sentry for error tracking. But I'm not getting error record of this error. It's native error. But I already gave trace path how to reproduce it. I will give you exact native error while debugging in android simulator asap. There are so many issues already created on this thread where they clearly mention native error. 
[https://github.com/react-native-community/react-native-camera/issues/357#issuecomment-353083409](url)

shahchaitanya, 2018-08-03:
Here it's stacktrace.

Here it's stacktrace.
`Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.reactnative.facedetector.RNFaceDetector.release()' on a null object reference
        at org.reactnative.camera.RNCameraView.onHostDestroy(RNCameraView.java:458)
        at com.facebook.react.bridge.ReactContext.onHostDestroy(ReactContext.java:230)
        at com.facebook.react.ReactInstanceManager.moveToBeforeCreateLifecycleState(ReactInstanceManager.java:690)
        at com.facebook.react.ReactInstanceManager.onHostDestroy(ReactInstanceManager.java:599)
        at com.facebook.react.ReactInstanceManager.onHostDestroy(ReactInstanceManager.java:613)
        at com.facebook.react.ReactActivityDelegate.onDestroy(ReactActivityDelegate.java:119)
        at com.facebook.react.ReactActivity.onDestroy(ReactActivity.java:72)
        at android.app.Activity.performDestroy(Activity.java:7209)
        at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1265)
        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4629)
        at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:4660) 
        at android.app.ActivityThread.-wrap5(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1749) 
        at android.os.Handler.dispatchMessage(Handler.java:105) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6809)`

SimonErm, 2018-09-25:
Fixed in https://github.com/react-native-community/react-native-camera/commit/71a535571e4d06d8368bd46987f3878831a14c07

closed 2018-09-25 by SimonErm