react-native-camera/react-native-camera #1729: Failed when Rotate with React-Navigation - setPreviewTexture failed

matamicen, 2018-08-05:
Hi guys,
We are having a crash in Android when we rotate the emulator when the camera is active, could be the react-navigator with redux? becasue we have the same project with the same versions of react-native-camera, compileSdkVersion, buildToolsVersion, etc and it is working without any problem (the working project doesn't have redux and react-navigation).

More info: The camera is on a StackNavigator Screen

it throws this error:
java.io.IOException: setPreviewTexture failed

setPreviewTexture failed
setUpPreview
Camera1.java:176
onSurfaceChanged
Camera1.java:109
dispatchSurfaceChanged
PreviewImpl.java:56
onSurfaceTextureSizeChanged
TextureViewPreview.java:53
onSizeChanged
TextureView.java:366
sizeChange
View.java:17641
setFrame
View.java:17603
layout
View.java:17520
onLayout
RNCameraView.java:191
layout
View.java:17523
layout
ViewGroup.java:5612
updateLayout
NativeViewHierarchyManager.java:241
updateLayout
NativeViewHierarchyManager.java:206
execute
UIViewOperationQueue.java:175
run
UIViewOperationQueue.java:894
flushPendingBatches
UIViewOperationQueue.java:1001
access$2400
UIViewOperationQueue.java:46
doFrameGuarded
UIViewOperationQueue.java:1061
doFrame
GuardedFrameCallback.java:29
doFrame
ReactChoreographer.java:134
doFrame
ChoreographerCompat.java:105
run
Choreographer.java:869
doCallbacks
Choreographer.java:683
doFrame
Choreographer.java:616
run
Choreographer.java:857
handleCallback
Handler.java:751
dispatchMessage
Handler.java:95
loop
Looper.java:154
main
ActivityThread.java:6077
invoke
Method.java
run
ZygoteInit.java:866
main
ZygoteInit.java:756

cactis, 2019-01-17:
``
java.lang.RuntimeException: java.io.IOException: setPreviewTexture failed
 at com.google.android.cameraview.Camera1.setUpPreview(Camera1.java:189)
 at com.google.android.cameraview.Camera1$1.onSurfaceChanged(Camera1.java:119)
 at com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:56)
 at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:46)
 at android.view.TextureView.getHardwareLayer(TextureView.java:387)
 at android.view.TextureView.draw(TextureView.java:325)
 at android.view.View.updateDisplayListIfDirty(View.java:16081)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at android.view.View.draw(View.java:17102)
 at android.view.View.updateDisplayListIfDirty(View.java:16081)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:699)
 at android.view.View.draw(View.java:17102)
 at android.view.View.updateDisplayListIfDirty(View.java:16081)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:699)
 at android.view.View.updateDisplayListIfDirty(View.java:16076)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:699)
 at android.view.View.updateDisplayListIfDirty(View.java:16076)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at android.view.View.updateDisplayListIfDirty(View.java:16076)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at android.view.View.draw(View.java:17102)
 at android.widget.HorizontalScrollView.draw(HorizontalScrollView.java:1643)
 at com.facebook.react.views.scroll.ReactHorizontalScrollView.draw(ReactHorizontalScrollView.java:415)
 at android.view.View.updateDisplayListIfDirty(View.java:16081)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:699)
 at android.view.View.updateDisplayListIfDirty(View.java:16076)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at android.view.View.draw(View.java:17102)
 at android.widget.ScrollView.draw(ScrollView.java:1722)
 at com.facebook.react.views.scroll.ReactScrollView.draw(ReactScrollView.java:392)
 at android.view.View.updateDisplayListIfDirty(View.java:16081)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:699)
 at android.view.View.draw(View.java:17102)
 at android.view.View.updateDisplayListIfDirty(View.java:16081)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3552)
 at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:699)
 at android.view.View.updateDisplayListIfDirty(View.java:16076)
 at android.view.View.draw(View.java:16865)
 at android.view.ViewGroup.drawChild(ViewGroup.java:3766)
 at android.view.ViewGroup.dispatchDraw(ViewGroup
``

"react-native-camera": "^1.8.0",

n1ru4l, 2019-01-17:
@cactis `RNCamera` has nothing to do with picking photos from an album. Can you please explain more what si causing this issue?

closed 2019-06-08 by stale[bot]

iOSQiao, 2019-07-04:
If you used an RNCamera on the previous page, you need to remove the Camera component from the previous page from the view before moving on to the next page to use an RNCamera. You can remove the Camera component from the previous page by setting setState.

prodoxx, 2019-09-17:
@iOSQiao, can you give an example?

iOSQiao, 2019-10-25:
let me see.

willymulyana, 2020-05-21:
Thanks @iOSQiao it works for me. My code is like this:

```
  takePicture = async() => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true, fixOrientation:true, forceUpOrientation:true };
      const data = await this.camera.takePictureAsync(options);
      this.setState({pictureTaken:true},()=>{
        // go to preview screen
        setTimeout(()=>{
          this.setState({pictureTaken:false});
        }, 500);
      })
    }
  };

// on render part
{
  this.state.pictureTaken ? null :
    <RNCamera
        ref={ref => {
          this.camera = ref;
        }}
        style={{width:'100%', height:'100%'}}
        type={RNCamera.Constants.Type.back}
        flashMode={RNCamera.Constants.FlashMode.off}
        androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
        }}
    />
}
```

iOSQiao, 2020-05-30:
Like this,

`this.didFocusSubscription = this.props.navigation.addListener('didFocus', (payload) => {
            this.setState({ cameraPreview: true });
        });`
`this.didBlurSubscription = this.props.navigation.addListener('didBlur', (payload) => {
            this.setState({ cameraPreview: false });
        });`