react-native-camera/react-native-camera #1733: Could not find method google() for arguments [] on repository container.

rmnrajan, 2018-08-07:
Where:
Build file 'C:\Users\user\Desktop\TP\Software\Chatbot_MobileApp\tmp\ChatbotIonicApp\platforms\android\build.gradle' line: 28

* What went wrong:
A problem occurred evaluating root project 'android'.
> Could not find method google() for arguments [] on repository container.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
(node:12616) UnhandledPromiseRejectionWarning: Error: cmd: Command failed with exit code 1 Error output:
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\user\Desktop\TP\Software\Chatbot_MobileApp\tmp\ChatbotIonicApp\platforms\android\build.gradle' line: 28

* What went wrong:
A problem occurred evaluating root project 'android'.
> Could not find method google() for arguments [] on repository container.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
    at ChildProcess.whenDone (C:\Users\user\Desktop\TP\Software\Chatbot_MobileApp\tmp\ChatbotIonicApp\platforms\android\cordova\node_modules\cordova-common\src\superspawn.js:169:23)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
(node:12616) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:12616) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
PS C:\Users\user\Desktop\TP\Software\Chatbot_MobileApp\tmp\ChatbotIonicApp> ionic cordova run android

sibelius, 2018-08-07:
check this https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md

closed 2018-08-07 by sibelius

tobyhinloopen, 2018-08-07:
That worked for me, thank you @sibelius