react-native-camera/react-native-camera #1734: on a null object reference

ShawSmall, 2018-08-07:
react-native-camera 1.1.5  android  
When i quit app, make a mistake。
java.lang.NullPointerException: Attempt to invoke virtual method 'void com.google.zxing.MultiFormatReader.reset()' on a null object reference
        at org.reactnative.camera.RNCameraView.onBarCodeScanningTaskCompleted(RNCameraView.java:295)
        at org.reactnative.camera.tasks.BarCodeScannerAsyncTask.onPostExecute(BarCodeScannerAsyncTask.java:71)
        at org.reactnative.camera.tasks.BarCodeScannerAsyncTask.onPostExecute(BarCodeScannerAsyncTask.java:10)
        at android.os.AsyncTask.finish(AsyncTask.java:651)
        at android.os.AsyncTask.access$500(AsyncTask.java:180)
        at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:668)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:171)
        at android.app.ActivityThread.main(ActivityThread.java:5729)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:732)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:622)

sibelius, 2018-08-07:
this looks like a bug on package com.google.zxing, could you open an issue there as well?

Symous, 2019-03-08:
@pavermakov Hi,friend, have you solve this? I have met the same error.

closed 2019-06-07 by stale[bot]