react-native-camera/react-native-camera #1735: undefined is not an object 'CameraManager.Aspect.Fill'

luisrodge, 2018-08-07:
react-native-camera 1.1.4 android

For some reason I keep getting the aforementioned error on android when running the app on my device.
There is no way to get rid of the error other than reinstalling the app entirely.

sibelius, 2018-08-07:
this looks like a linking issue

luisrodge, 2018-08-07:
I have done everything according to the documentation.

patt-makeitsimple, 2018-08-21:
Have you console.log(CameraModule) in Camera.js? 
I have the same issue, this will get error randomly. Because I have also installed react-native-camera-kit, I try to add a new function in react-native-camera-kit CameraModule.java , e.g.
    
@ReactMethod
public void cameraKitByTrial() {

} 

After clean gradle and run again, I found the cameraKitByTrial function will be printed out in console.log . That mean the CameraModule loaded in Camera.js is the module from react-native-camera-kit. This is because their module name CameraModule are the same. 

To solve it, I rename the module name in RCTCamera.java getName method to RCTCameraModuleDeprecated and change the NativeModules.CameraModule to NativeModules.CameraModuleDeprecated. The error then gone.

SimonErm, 2018-11-20:
can this be closed?

sibelius, 2018-11-20:
relink properly

closed 2018-11-20 by sibelius

olanMbakop, 2019-04-27:
> Have you console.log(CameraModule) in Camera.js?
> I have the same issue, this will get error randomly. Because I have also installed react-native-camera-kit, I try to add a new function in react-native-camera-kit CameraModule.java , e.g.
> 
> @ReactMethod
> public void cameraKitByTrial() {
> 
> }
> 
> After clean gradle and run again, I found the cameraKitByTrial function will be printed out in console.log . That mean the CameraModule loaded in Camera.js is the module from react-native-camera-kit. This is because their module name CameraModule are the same.
> 
> To solve it, I rename the module name in RCTCamera.java getName method to RCTCameraModuleDeprecated and change the NativeModules.CameraModule to NativeModules.CameraModuleDeprecated in Camera.js. The error then gone.

Hi, I have the same issue. Can you elaborate more on the procedure you used? ie. Steps.

I'm stuck in here `RCTCamera.java getName`