react-native-camera/react-native-camera #1736: undefined is not an object (evaluating 'CameraManager.Aspect') #ios #bug

cartesv, 2018-08-07:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

cartesv, 2018-08-07:
In Camera.js, classes NativeModules.CameraManager || NativeModules.CameraModule are not defined.

keske, 2018-08-09:
@cartesv Look this: https://github.com/react-native-community/react-native-camera/issues/1407#issuecomment-392881900

I had the same issue and that script helps me.

fantasyStyle, 2018-09-03:
make sure you have linked xcode's linked frameworks and librales with libRNCamera.a And project finder existed the rncamera.xcodeproj

ashavit, 2018-09-26:
It didn't for me either. I think the 5 steps of installation are not enough.
I've noticed that RCTCameraManager.m (for iOS) has "Aspect" returned as part of it's constants (constantsToExport method).
But the same method in RNCameraManager,m does not have it. it is missing other constants as well.
is it on purpose? or could this be the reason?

albullington, 2018-10-09:
I'm stuck on the same error. I tried both react-native link for automatic installation, and I tried the 5 manual steps of installation. Haven't found a solution.

pawanacharya, 2018-10-09:
Stuck with the same error. The manual linking doesn't solve my issue either.

gmaci005, 2018-10-11:
I'm stuck on the same error.

GerardoLucero, 2018-10-15:
I'm stuck on the same error X2

albullington, 2018-10-15:
I ended up solving by shutting down my simulators and devices, then re-running npm start and react-native run-ios/run-android. Maybe linking the camera library requires a rebuild?

xcytek, 2018-11-20:
Also, in Xcode under project navigation in the Info.plist file, you have to specify the Privacy - Camera Usage Description with a string Value. Otherwise, you will get an error.

farkhanda44zeb, 2018-12-06:
same issue:(

sibelius, 2018-12-06:
link is wrong

relink

closed 2018-12-06 by sibelius

prakhar-goel, 2019-01-16:
Try `react-native link`