react-native-camera/react-native-camera #1737: evaluating h.Aspect.fill

dileepbolisetti, 2018-08-08:
### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",

### Steps to reproduce
1. npm i
2. react-native run-android

### Expected behaviour

### Actual behaviour
 -if you force stop the application sometimes working
-even in production apk is crashing 3-4 times and then starts working after force stop the app works fine for sometimes..

### Environment
- **React Native version**:
"react-native": "0.55.4",
 "react": "16.3.1"

- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Android 7.1.1

### react-native-camera
**Version**: npm version or "master"
master 

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

com.facebook.react.modules.core.ExceptionsManagerModule.showOrThrowError(ExceptionsManagerModule.java:54)
       at com.facebook.react.modules.core.ExceptionsManagerModule.reportFatalException(ExceptionsManagerModule.java:38)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
       at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:160)
       at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
       at android.os.Looper.loop(Looper.java:154)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
       at java.lang.Thread.run(Thread.java:761)

--

Fatal Exception: com.facebook.react.common.JavascriptException: undefined is not an object (evaluating 'h.Aspect.fill'), stack:
<unknown>@613:6802
d@2:768
n@2:409
t@2:262
<unknown>@612:114
d@2:768
n@2:409
t@2:262
<unknown>@576:544
d@2:768
n@2:409
t@2:262
<unknown>@311:310
d@2:768
n@2:409
t@2:262
<unknown>@310:158
d@2:768
n@2:409
t@2:262
<unknown>@307:158
d@2:768
n@2:409
t@2:262
<unknown>@306:81
d@2:768
n@2:409
t@2:262
<unknown>@12:42
d@2:768
n@2:339
t@2:262
global code@850:8

       at com.facebook.react.modules.core.ExceptionsManagerModule.showOrThrowError(ExceptionsManagerModule.java:54)
       at com.facebook.react.modules.core.ExceptionsManagerModule.reportFatalException(ExceptionsManagerModule.java:38)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
       at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:160)
       at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
       at android.os.Looper.loop(Looper.java:154)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
       at java.lang.Thread.run(Thread.java:761)

jgcmarins, 2018-08-08:
@dileepbolisetti is there a repo to reproduce this?

dileepbolisetti, 2018-08-08:
@jgcmarins created a repo with sample code 
https://github.com/dileepbolisetti/react-native-camera-test

and 

https://github.com/dileepbolisetti/react-native-camera-test/tree/master/android/app/release
app-release.apk is behaving as i mentioned earlier..

(y)

dileepbolisetti, 2018-08-10:
hi @jgcmarins did u get the chance to see the  issue !!

wolfgio, 2018-10-10:
Any update on this?

sibelius, 2019-04-18:
your link is wrong

closed 2019-04-18 by sibelius