react-native-camera/react-native-camera #1741: Path option does not continue recording video over the same file on iOS

halilb, 2018-08-10:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
I am trying to create an Instagram-like video recorder, and trying to make use of this explanation of RNCamera for that purpose:

> path (file path on disk). Specifies the path on disk to record the video to. You can use the same uri returned to continue recording across start/stops

So I am storing the file _uri_ after the first record, and passing that _uri_ as the **path** option to **recordAsync** method. Here is all the logic:

 ```
state = {
  recordUri: null,
};

startCapturing = () => 
   console.log('passed path', this.state.recordUri);

  this.cameraRef
    .recordAsync({
      path: this.state.recordUri,
    }).then(data => {
       console.log('saved path', data.uri);

      if (!this.state.recordUri) {
        this.setState({
          recordUri: data.uri,
        });
      }
    });
}

stopCapturing = () => {
  this.cameraRef.stopRecording();
};
```

And after invoking _startCapturing_ and _stopCapturing_  methods 6 times(waiting 2 seconds before making the next method call), here is the console output:

> passed path null
> saved path file:///var/mobile/Containers/Data/Application/DC7149D5-4FA6-4DBB-BF3E-BBC97CD106C8/Library/CachesCamera/B5069A3B-60EC-4AE0-A308-722D92F2EF77.mov
> passed path file:///var/mobile/Containers/Data/Application/DC7149D5-4FA6-4DBB-BF3E-BBC97CD106C8/Library/CachesCamera/B5069A3B-60EC-4AE0-A308-722D92F2EF77.mov
> saved path file:///var/mobile/Containers/Data/Application/DC7149D5-4FA6-4DBB-BF3E-BBC97CD106C8/Library/CachesCamera/5E4CC555-CF28-4512-A129-35039617AB2A.mov
> passed path file:///var/mobile/Containers/Data/Application/DC7149D5-4FA6-4DBB-BF3E-BBC97CD106C8/Library/CachesCamera/B5069A3B-60EC-4AE0-A308-722D92F2EF77.mov
> saved path file:///var/mobile/Containers/Data/Application/DC7149D5-4FA6-4DBB-BF3E-BBC97CD106C8/Library/CachesCamera/3DF6FF8C-A8A6-4D7C-98F0-E756CEC1EC02.mov

### Expected behaviour
The video record should have been appended to the first file, that is B5069A3B-60EC-4AE0-A308-722D92F2EF77.mov.

### Actual behaviour
Even though I passed _B5069A3B-60EC-4AE0-A308-722D92F2EF77.mov_ path to recordAsync method, it created a new video file. And it kept creating separate video files for each recordAsync calls. _B5069A3B-60EC-4AE0-A308-722D92F2EF77.mov_ file contains only the video captured until the first stopRecording call.

### Environment
- **React Native version**: 0.55.3
- **React Native platform + platform version**: iOS 11.0

### react-native-camera
**Version**: 1.2.0

Note: I also tried passing a predefined file path using react-native-fs library. That also produced the same problem.

SimonErm, 2018-09-17:
#1754

georgetmayorga, 2018-11-01:
@halilb Wondering if this was resolved? The tagged issue doesn't seem to be helpful. 

I'm also passing in the `path` option to `recordAsync`, recording a few clips but the behavior I'm getting is that only the last clip comes out—it seems to be overwriting rather than merging the clips.

I'm on RNCamera 1.3.1 on iOs

halilb, 2018-11-01:
Nope @georgetmayorga. I couldn't make it work.

kgrajan12, 2018-12-15:
i'm also stopped in this same problem

n1ru4l, 2018-12-18:
@halilb  @kgrajan12 AFAIK the `path` option was never intended to append the recording to an already existing file. However since this seems like a nice feature I encourage you all to implement it and send a pull request.

michaelfreund, 2019-12-15:
#2507

michaelfreund, 2019-12-17:
@sibelius @n1ru4l Can you guys give us some brief information on the status of this issue?

matiborr, 2020-04-03:
Does anyone have a solution to this problem?

n1ru4l, 2020-04-03:
@matiborr As there is no solution mentioned here I encourage you to implement it yourself and send a pull request. react-native-camera lives of its community contributions.

Please try to avoid spamming issues with `Asking for status`-questions, as it results in unnecessary noise for the maintainers 😅.

matiborr, 2020-04-03:
How are you? Could you tutor me to implement it?
I have found in Camera.tsx "CameraManager.pausePreview" but I cannot find how to pause the recording and then resume it

closed 2020-06-09 by stale[bot]