react-native-camera/react-native-camera #1743: camera preview rotates when there is no need

PavelPolyakov, 2018-08-13:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
Here is the repo:
https://github.com/PavelPolyakov/react-native-camera-rotation-bug
the `orientation` must be fixed to the `portrait`:
https://github.com/PavelPolyakov/react-native-camera-rotation-bug/blob/master/app.json#L10

1. Run the app
2. Turn the phone to the landscape mode

### Does it work with Expo Camera?
the bug was discovered using `expo`

### Expected behaviour
Image preview does not turn, it just continues to show what I see.

### Actual behaviour
Image preview is turned `90deg` and it shows not what I actually see.

Regards,

blakesanie, 2018-09-13:
No idea what's happening. Same bug here!

sibelius, 2018-09-13:
try with master

it should be fixed by https://github.com/react-native-community/react-native-camera/pull/1786

SimonErm, 2018-09-26:
Is this a Android or iOS issue?

n1ru4l, 2018-12-12:
Closing due to missing info and multiple other similar issues being open (eg #1725).

closed 2018-12-12 by n1ru4l