react-native-camera/react-native-camera #1744: react-native-camera check permissions before using RNCamera

tonmanayo, 2018-08-13:
using RNCamera

There is no way to check the permissions status on IOS before actually trying to use the RNCamera

I should be able to check IOS camera permissions before using the camera
- **React Native version**: Latest

### react-native-camera
**Version**: npm version or "master" latest

n1ru4l, 2018-08-15:
You might use https://github.com/yonahforst/react-native-permissions for this.

A PR that exposes a method on `RNCamera` would also be handy.

tonmanayo, 2018-08-15:
@n1ru4l thanks for the response. I really don't want to use that library at the moment as one of the issues says that it gets rejected by the apple app store, and the camera is the only lib that I need the permissions for.

n1ru4l, 2018-08-15:
@tonmanayo Well we have an app in the app store that uses this library.

sibelius, 2018-08-15:
this should be in user land, or in another library

closed 2018-08-16 by n1ru4l

n1ru4l, 2018-12-20:
@tonmanayo We are open to pull requests that expose methods for asking for the desired permissions.

haspindergill, 2021-03-02:
Why this issue is closed, I didn't find any satisfactory answer. How we can ask for camera permissions without adding RNCamera component?

MateusAndrade, 2021-03-02:
> Why this issue is closed, I didn't find any satisfactory answer. How we can ask for camera permissions without adding RNCamera component?

Hi @haspindergill, I think you can try using [react-native-permissions](https://github.com/zoontek/react-native-permissions). Right now, I don't think you will be able to use the permission  API directly from RNCamera without mounting the component.