react-native-camera/react-native-camera #1745: barcode scanner cant recognize small size barcodes (html size:  200px*100px). but big ones are ok

veeyka, 2018-08-15:
its working fine on ios.
but it dont recognize small barcodes 
i'm using code128 
300*300 px barcodes are identifying goodly

                <RNCamera
                    ref={ref => {
                        this.camera = ref;
                    }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                    flashMode={RNCamera.Constants.FlashMode.on}
                    permissionDialogTitle={'Permission to use camera'}
                    permissionDialogMessage={'We need your permission to use your camera phone'}
                    barCodeTypes={[RNCamera.Constants.BarCodeType.code128]}
                    onBarCodeRead={this.onBarCodeRead.bind(this)} />

agrass, 2019-03-18:
Same problem using GoogleVisionBarcode, trying to scan a small QR without results. When I try a big QR it works perfect.

sibelius, 2019-03-18:
does it work with MLKit on v2?

agrass, 2019-03-18:
I'm using v2.0.1 and still having problems when reading small QR codes.  Is a QR of 0.8 inches approx. I tried with other apps and they read it very fast without problems. Any recommendation about what I can try or change to improve this?

agrass, 2019-03-18:
I'm testing this in Android

closed 2019-06-07 by stale[bot]

josectobar, 2019-12-05:
I'm having the same issue when trying to scan a small 128 barcode using iPhones X or 11/11 Pro. It seems that the scanner has a hard time detecting the right barcode, it seems to work better with bigger barcodes from a different kind.