react-native-camera/react-native-camera #1747: React Pod has been deprecated and causes conflict with node_modules/react-native

NordlingArt, 2018-08-16:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. `yarn add react-native-camera`
2. `react-native link react-native-camera`
3. Podspec gets added to Podfile but when running the project, I see no `CameraManager` inside `NativeModules`.
4. I manually run a `pod install` in order for it to get added.
5. Message I get after pods are installed is: "[!] React has been deprecated"
6. I look inside the Podspec of react-native-camera and see that `React` Pod is specified as a dependency, which conflicts with node_modules/react-native. They share the same name in their package.json: "react-native". Message when running the project:

```
Loading dependency graph...(node:73752) UnhandledPromiseRejectionWarning: Error: jest-haste-map: @providesModule naming collision:
  Duplicate module name: react-native
  Paths: ~/Apps/CameraApp/ios/Pods/React/package.json collides with ~/Apps/CameraApp/node_modules/react-native/package.json
```

### Expected behaviour
Everything should be implemented after `yarn add react-native-camera` and then `react-native link -react-native-camera`.

### Actual behaviour
I need to do a manual `pod install` because the linking method never implements `CameraManager`. Another error occurs instead due to `React` Pod being deprecated and causing a conflict.

### Environment
- **React Native version**: 0.56.0
- **React Native platform + platform version**: iOS 11.4

### react-native-camera
**Version**: 1.2.0

onmyway133, 2018-08-29:
@NordlingArt I installed manually like you described, still the issue

yli18, 2018-09-13:
react-native link works for me, but i had the similar error when i tried CocoaPods install.

I was playing with the latest ios textRecognition function. Since CocoaPods Path is the only install option if i want to include Text Recognition function.  

when pod install it has [!] React has been deprecated warning
compile fail because of the 'React/RCTBridge.h' file not found error, also with the following warning from console

This warning is caused by a @providesModule declaration with the same name across two different files.
jest-haste-map: @providesModule naming collision:
  Duplicate module name: Array.prototype.es6
  Paths: /Users/ying/WorkSpace/crna/ios/Pods/React/packager/react-packager/src/DependencyResolver/polyfills/Array.prototype.es6.js collides with /Users/ying/WorkSpace/crna/node_modules/react-native/Libraries/polyfills/Array.prototype.es6.js

This warning is caused by a @providesModule declaration with the same name acros

n1ru4l, 2018-09-24:
Please read the documentation and checkout the Podfile setup guide of react-native http://facebook.github.io/react-native/docs/integration-with-existing-apps.

Never use the deprecated react pod which is published to cocoapods.

closed 2018-09-24 by n1ru4l

n1ru4l, 2018-11-16:
@jqn Please send a pull request if you think we could improve the documentation.