react-native-camera/react-native-camera #1748: Incompatible with React Native 0.57 (RC)

haggholm, 2018-08-17:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Upgrade to `react-native@next`
2. Attempt to build the Android client

### Expected behaviour
Successful compilation

### Actual behaviour
```
> Task :react-native-camera:compileDebugJavaWithJavac
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:7: error: cannot find symbol
import com.facebook.react.uimanager.UIBlock;
                                   ^
  symbol:   class UIBlock
  location: package com.facebook.react.uimanager
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/CameraReadyEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarcodesDetectedEvent.java:40: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/PictureSavedEvent.java:27: error: method init in class Event<T> cannot be applied to given types;
        super.init(viewTag);
             ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FacesDetectedEvent.java:50: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FaceDetectionErrorEvent.java:28: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/CameraMountErrorEvent.java:27: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarcodeDetectionErrorEvent.java:29: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/TextRecognizedEvent.java:55: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarCodeReadEvent.java:34: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:189: error: cannot find symbol
        uiManager.addUIBlock(new UIBlock() {
                                 ^
  symbol:   class UIBlock
  location: class CameraModule
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:210: error: cannot find symbol
        uiManager.addUIBlock(new UIBlock() {
                                 ^
  symbol:   class UIBlock
  location: class CameraModule
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:232: error: cannot find symbol
    uiManager.addUIBlock(new UIBlock() {
                             ^
  symbol:   class UIBlock
  location: class CameraModule
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:255: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:278: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:299: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
/mnt/scratch/shared/petter/witkit-devenv/client-react-native-navigation/root/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:325: error: cannot find symbol
        uiManager.addUIBlock(new UIBlock() {
                                 ^
  symbol:   class UIBlock
  location: class CameraModule
```

### Environment
- **React Native version**: 0.57.0-rc.0
- **React Native platform + platform version**: Android (platform version irrelevant)

### react-native-camera
**Version**: 1.2.0

wynch, 2018-09-04:
Same problem here, with React Native 0.57.0-rc.4

SimonErm, 2018-09-22:
Did you try 0.57.0?

wynch, 2018-09-22:
Sure, it's alright with RN 0.57.0 :)

sibelius, 2018-10-02:
working fine on RN57 https://github.com/react-native-community/rncamera-example/pull/13

closed 2018-10-02 by sibelius

Michaelvilleneuve, 2019-01-02:
Same problem on 0.57.8
Also, it does not seem to happen on every machine since our build machine keeps working

balthazar, 2019-01-06:
Same, had to downgrade to `0.57.7`

furedal, 2019-01-07:
This should probably be reopened, ```RN 0.57.8``` breaks the latest version of RNC.
For now, it works using ```RN 0.57.7```

This is what I am getting, looks like the original issue
`/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:7: error: cannot find symbol
import com.facebook.react.uimanager.UIBlock;
                                   ^
  symbol:   class UIBlock
  location: package com.facebook.react.uimanager
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:213: error: cannot find symbol
        uiManager.addUIBlock(new UIBlock() {
                                 ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:234: error: cannot find symbol
        uiManager.addUIBlock(new UIBlock() {
                                 ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:256: error: cannot find symbol
    uiManager.addUIBlock(new UIBlock() {
                             ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:279: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:302: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:323: error: cannot find symbol
      uiManager.addUIBlock(new UIBlock() {
                               ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:349: error: cannot find symbol
        uiManager.addUIBlock(new UIBlock() {
                                 ^
  symbol:   class UIBlock
  location: class CameraModule
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarcodeDetectionErrorEvent.java:29: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarCodeReadEvent.java:37: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarcodesDetectedEvent.java:40: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/CameraMountErrorEvent.java:27: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
[... 34 lines cut ...]
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/PictureTakenEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/TextRecognizedEvent.java:55: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
18 errors`

sibelius, 2019-01-07:
this happens when you resolve react-native wrong on gradle

dap1995, 2019-02-07:
I am trying to update to 0.58.4, and I'm getting the same error. How to solve that?

> This should probably be reopened, `RN 0.57.8` breaks the latest version of RNC.
> For now, it works using `RN 0.57.7`
> 
> This is what I am getting, looks like the original issue
> 
> ```
> import com.facebook.react.uimanager.UIBlock;
>                                    ^
>   symbol:   class UIBlock
>   location: package com.facebook.react.uimanager
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:213: error: cannot find symbol
>         uiManager.addUIBlock(new UIBlock() {
>                                  ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:234: error: cannot find symbol
>         uiManager.addUIBlock(new UIBlock() {
>                                  ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:256: error: cannot find symbol
>     uiManager.addUIBlock(new UIBlock() {
>                              ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:279: error: cannot find symbol
>       uiManager.addUIBlock(new UIBlock() {
>                                ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:302: error: cannot find symbol
>       uiManager.addUIBlock(new UIBlock() {
>                                ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:323: error: cannot find symbol
>       uiManager.addUIBlock(new UIBlock() {
>                                ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:349: error: cannot find symbol
>         uiManager.addUIBlock(new UIBlock() {
>                                  ^
>   symbol:   class UIBlock
>   location: class CameraModule
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarcodeDetectionErrorEvent.java:29: error: method init in class Event<T> cannot be applied to given types;
>     super.init(viewTag);
>          ^
>   required: int,long
>   found: int
>   reason: actual and formal argument lists differ in length
>   where T is a type-variable:
>     T extends Event declared in class Event
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarCodeReadEvent.java:37: error: method init in class Event<T> cannot be applied to given types;
>     super.init(viewTag);
>          ^
>   required: int,long
>   found: int
>   reason: actual and formal argument lists differ in length
>   where T is a type-variable:
>     T extends Event declared in class Event
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarcodesDetectedEvent.java:40: error: method init in class Event<T> cannot be applied to given types;
>     super.init(viewTag);
>          ^
>   required: int,long
>   found: int
>   reason: actual and formal argument lists differ in length
>   where T is a type-variable:
[... 38 lines cut ...]
>   reason: actual and formal argument lists differ in length
>   where T is a type-variable:
>     T extends Event declared in class Event
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/PictureTakenEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
>     event.init(viewTag);
>          ^
>   required: int,long
>   found: int
>   reason: actual and formal argument lists differ in length
>   where T is a type-variable:
>     T extends Event declared in class Event
> /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/TextRecognizedEvent.java:55: error: method init in class Event<T> cannot be applied to given types;
>     super.init(viewTag);
>          ^
>   required: int,long
>   found: int
>   reason: actual and formal argument lists differ in length
>   where T is a type-variable:
>     T extends Event declared in class Event
> Note: Some input files use or override a deprecated API.
> Note: Recompile with -Xlint:deprecation for details.
> Note: /Users/Chrille/Documents/my_app/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java uses unchecked or unsafe operations.
> Note: Recompile with -Xlint:unchecked for details.
> 18 errors```
> ```