react-native-camera/react-native-camera #1752: black screeen in android

swaroopa94, 2018-08-22:
What's happening?
Hi,
I installed this module and linked the modules as given in description but I am getting a black screen in android the camera is not opening up.

Build details?
android version 7.1.1
react-native-qrcode-scanner - ^1.0.1
react-native-camera - ^1.1.4

Here are my build.gradle

`buildscript {
repositories {
google()
maven { url "https://maven.google.com" }
jcenter()
}
dependencies {
classpath 'com.android.tools.build:gradle:3.1.0'

    // NOTE: Do not place your application dependencies here; they belong
    // in the individual module build.gradle files
}
}

allprojects {
repositories {

    maven { url "https://jitpack.io" }
     maven { url "https://maven.google.com" }
     google()
    mavenLocal()
    jcenter()
    maven {
        // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
        url "$rootDir/../node_modules/react-native/android"
    }
}
}

/**

Project-wide gradle configuration properties for use by all modules
*/
ext {
compileSdkVersion = 26
targetSdkVersion = 26
buildToolsVersion = "26.0.2"
googlePlayServicesVersion = "12.0.1"
supportLibVersion = "26.1.0"
}`

app/build.gradle

`dependencies {
compile (project(':react-native-camera')) {
exclude group: "com.google.android.gms"

compile ('com.google.android.gms:play-services-vision:12.0.1') {
    force = true
}
}
compile project(':react-native-google-analytics-bridge')
compile project(':react-native-vector-icons')
compile project(':react-native-video')
compile project(':react-native-push-notification')
compile project(':react-native-orientation')
compile project(':react-native-image-resizer')
compile project(':react-native-android-permissions')
compile project(':react-native-webrtc')
compile(project(':react-native-maps')){
exclude group: 'com.google.android.gms', module: 'play-services-base'
exclude group: 'com.google.android.gms', module: 'play-services-maps'
}

compile project(':react-native-device-info')
compile fileTree(dir: "libs", include: ["*.jar"])
compile "com.android.support:appcompat-v7:23.0.1"
compile "com.facebook.react:react-native:+"  // From node_modules
compile project(':react-native-image-picker')
   compile 'com.google.android.gms:play-services-base:+'
compile 'com.google.android.gms:play-services-maps:+'
compile fileTree(dir: "node_modules/react-native-device-info/android/libs", include: ["*.jar"])
}
`

in gradle.properties

distributionUrl=https://services.gradle.org/distributions/gradle-4.4-all.zip

Please anyone suggest me how to solve this

dileepbolisetti, 2018-08-28:
help us with this issue

SimonErm, 2018-09-22:
can you post the code you use with the camera?

basanthalluri, 2018-09-24:
Hi @SimonErm I have used the camera module for barcode scanner and here is what I kept for RNCamera

`<RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style = {styles.preview}
            type={RNCamera.Constants.Type.back}
            onBarCodeRead={this.barcodeReceived.bind(this)}
            flashMode={RNCamera.Constants.FlashMode.on}
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
        />`

It is working fine in iOS but in android device it showing a black screen and no log is there to verify.

swaroopa94, 2018-09-24:
Hi @SimonErm I have used the camera module for barcode scanner and here is what I kept for RNCamera

`<RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style = {styles.preview}
            type={RNCamera.Constants.Type.back}
            onBarCodeRead={this.barcodeReceived.bind(this)}
            flashMode={RNCamera.Constants.FlashMode.on}
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
        />`

It is working fine in iOS but in android device it showing a black screen and no log is there to verify.

SimonErm, 2018-09-25:
Did the permission Dialog pop up? Maybe it's a problem our permission request. Can you take a look at Settings->Apps->Application List->${YOUR_APP}->Permissions, if all permissions are granted? Furthermore did you add the permissions to your manifest?

pavermakov, 2018-10-10:
@swaroopa94 Do you have `<uses-permission android:name="android.permission.CAMERA" />` set in your `AndroidManifest.xml` file?

basanthalluri, 2018-10-10:
@pavermakov  Yes I added that line in my xml file

swaroopa94, 2018-10-10:
@pavermakov  yes I have added that line in my xml file

pavermakov, 2018-10-10:
@swaroopa94 I believe you need to check permission to use camera. I had the same issue when my targetSdkVersion was 22 and camera worked perfectly, but later I switched to 23 and got a black screen. I used  [react-native-permissions](https://github.com/yonahforst/react-native-permissions) to prompt for camera permissions and that's it! I had it working like before.

pavermakov, 2018-10-11:
@swaroopa94 You might also want to check permissions for storage, otherwise you might run into an issue like [this](https://github.com/react-native-community/react-native-camera/issues/1845).

miracle2k, 2018-10-26:
From what I can tell, react-native-camera itself should take care of asking for the permissions. It seems strange that asking for permissions manually should make any difference.

alamothe, 2019-01-05:
Looks like it's necessary to ask for permissions on Android. `react-native-camera` won't do this by itself, while on iOS it does.

n1ru4l, 2019-01-08:
This issue was fixed in https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.6

Please try to upgrade to the newest version.

closed 2019-01-08 by n1ru4l