react-native-camera/react-native-camera #1753: Expected camera component error when pausePreview is called before takePictureAsync on android

mshibl, 2018-08-22:
### Which implementation are you using
RNCamera - **android** (works on iOS)

### Steps to reproduce
1. Call camera.pausePreview()
2. Call camera.takePictureAsync()
3. catch and log error: `Expected camera component`

```    
takePicture = async () => {
        try {
            if (this.camera) {
                this.camera.pausePreview();

                const data = await this.camera.takePictureAsync();

                // ... use picture data
            }
        } catch (err) {
            console.error(err);
        }
    };
```

### Expected behaviour
The camera should pause the preview, and then take the picture 

### Actual behaviour
Throws an error: `Error: takePictureAsync: Expected a Camera component`

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: 1.2.0

staafl, 2018-09-03:
Same scenario, same error - Android 8, react-native-camera 1.2.0

inventsys, 2019-01-31:
Same problem here. Any solution?

niv-breez, 2019-05-12:
happen to me also, does anyone found a fix ?

winadiw, 2019-06-10:
happened to me.. Any fix?

closed 2019-08-16 by stale[bot]

FilippoArgenti, 2019-09-04:
Can confirm this is still happening on v3.3.0
When I call .takePictureAsync without .pausePreview first it works.

Nervado, 2019-11-13:
Same here:

RNCamera: 3.9.0
RN: 0.61.4

Plattaform: Android

` async function takePicture() {
    try {
      if (camera) {
        // Pause the camera preview

        // Update the state to indicate loading
        setLoading(true);

        // Set the options for the camera
        const options = {
          base64: true,
        };

        // Get the base64 version of the image
        camera.pausePreview();
        const data = await camera.takePictureAsync(options);

        // Call the identify function
        identifyImage(data.base64);
      }
    } catch (error) {
      Alert.alert('Something whent wrong! =(');
    }
}`

Nervado, 2019-11-13:
Same here:

RN: 0.61.4
RNCamera: 3.9.0
Plattaform: Android 

`` async function takePicture() {
    try {
      if (camera) {
        // Pause the camera preview

        // Update the state to indicate loading
        setLoading(true);

        // Set the options for the camera
        const options = {
          base64: true,
        };

        // Get the base64 version of the image
        camera.pausePreview();
        const data = await camera.takePictureAsync(options);

        // Call the identify function
        identifyImage(data.base64);
      }
    } catch (error) {
      Alert.alert('Something whent wrong! =(');
    }
  }
``

TnRc, 2020-07-10:
Has this been resolved?