react-native-camera/react-native-camera #1754: The first video got a uri, but it crashed the second time. Can you help me?

L-Mno, 2018-08-23:
*RNCamera

### Steps to reproduce

I implemented the camera according to the document and ran it perfectly on Android, but I had a problem with ios video. I got the uri I needed for the first video, and then it crashed when I did the second video.

        <RNCamera
               ref={ref => {
                      this.camera = ref;
               }}
               style={styles.preview}
               type={RNCamera.Constants.Type.back}
               captureAudio = {true}
               flashMode={RNCamera.Constants.FlashMode.on}
               permissionDialogTitle={'Permission to use camera'}
               permissionDialogMessage={'We need your permission to use your camera phone'}
        />

        startRecording = async function () {
           if (this.camera) {
              const options = { mute:false,};
              this.camera.recordAsync(options)
                .then(data => {
                        console.log(data['uri']);
                        let recordPromise = CameraRoll.saveToCameraRoll(data['uri']);
                        recordPromise.then(function (result) {
                            console.log('Video saved successfully！');
                        })
                .catch(error => {
                            console.log(error);
                        });
              });
          }
        

        }
        stopRecording = async function () {
          if (this.camera) {
              this.camera.stopRecording();
          }
       };
   
### Expected behaviour
You can record multiple times

### Actual behaviour
App crashes with the following error:

[image]

2018-08-23 09:51:21.222992+0800 SecurityAlert0704[3048:2812095] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMovieFileOutput startRecordingToOutputFileURL:recordingDelegate:] No active/enabled connections'
*** First throw call stack:
(0x182592d8c 0x18174c5ec 0x18808c174 0x1028640b8 0x1032051dc 0x10320519c 0x103213dfc 0x1032086ac 0x103214d54 0x10321be38 0x1821b7e70 0x1821b7b08)
libc++abi.dylib: terminating with uncaught exception of type NSException

React Native version: 0.54.2
React Native platform + platform version: iOS 9.4
react-native-camera
Version: "master"

ingviso, 2018-08-27:
For me release 1.1.2 works all releases after that give the same error

okilimnik, 2018-08-27:
the same error:
iphone X
IOS 11.4.1
RN 0.55.4
react-native-camera  1.2.0

SpQuyt, 2018-08-28:
@ingviso it worksssssssssss!!!
1.1.2 is good bro!

ooitakayuki, 2018-09-06:
i have same problem.
IOS 11.0
RN 0.55.4
react-native-camera 1.2.0

thanks @ingviso

rptwsthi, 2018-09-07:
I am facing same issue.

pribeh, 2018-09-10:
Same here. anybody have any insight into this?

jonathangreco, 2018-09-19:
I've the same issue.

I tried this code, wich works fine on ANDROID :

```
  recordVideo = () => {
    if (this.camera) {
      let duration = this.props.localScene.duration;
      this.setState({isRecording: true});
      this.counterInterval = setInterval(() => {
        if (this.state.isRecording) {
          if ((this.state.counter - 1) > 0) {
            this.setState({
              counterBottom: 0,
              counter: (this.state.counter - 1)
            });
          } else {
            this.setState({
              counter: I18n.t("action_text").toUpperCase()
            });
            // Show the overlay
            clearInterval(this.counterInterval);

            this.camera.recordAsync({
              maxDuration: duration
            }).then(data => {
              CameraRoll.saveToCameraRoll(data.uri).then((uri) => {
                console.log(uri);
              });
              this.setState({
                isRecording: false,
                counter: 5,
              });
            }).catch(() => {
              this.stopRecording();
            });

            this.counterBottomInterval = setInterval(() => {
              if ((this.state.bottomCounter + 1) <= duration) {
                this.setState({
                  counter: "",
                  bottomCounter: (this.state.bottomCounter + 1)
                });
              } else {
                // Show the overlay
                this.setState({
                  counter: I18n.t("done_text").toUpperCase(),
                  bottomCounter: 0
                });
                clearInterval(this.counterBottomInterval);
              }
            }, 1000);
          }
        }
      }, 1000);
    }
  };

```
With an IOS device version 9.3.5 and react-native-camera 1.2.0 I've the error from this issue.
With the same device and react-native-camera 1.1.2 I've the #1550 issue...

What should I do ?
How can I run React-native-camera  wainting this issue to be fixed ?

MrPluto, 2018-09-19:
@rptwsthi maybe related #1801

webmasterdragon, 2018-09-22:
1.1.2 is working fine on iOS and 1.2.0 is working fine on Android.

darekg11, 2018-10-17:
1.1.2 is the only version that is working for me on both IOS and Android with multiple recordings. 1.1.3 is failing as soon as I want to record once more. Is there possibility for someone with more IOS knowledge bisect commits between those two? By looking at changelog included in 1.1.3:
https://github.com/react-native-community/react-native-camera/commit/98b1895038ccf47f94d7d27811f3540d3847feb7
There is a mention of:  
```**cache:**  store video recordings in same directory as photos ([bba84a98](https://github.com/react-native-community/react-native-camera/commit/bba84a983446c25f76aa77793f49d4252cd63ea3))```:
https://github.com/react-native-community/react-native-camera/commit/bba84a983446c25f76aa77793f49d4252cd63ea3
Can something like this break it on IOS? 🤔

closed 2019-08-07 by stale[bot]

fabriziobertoglio1987, 2019-10-18:
I am experiencing this issue on an iPhone 5 running ios 10.3.4. I get error `Thread 40: signal SIGABRT` every time I call `this.camera.recordAsync()`
The device is very old and I hopefully I'll purchase a new one and hope to not experience this issue.

KevRyan2, 2021-03-12:
Looks like this was never solved. After the first recording is completed, if you try a second recording before un-mounting/re-mounting the camera "Video is not recording" is thrown.  This is because it never starts recording. Further inspection: I believe after the first recording self.movieFileOutput is set to nil, preventing it from recording again at the expression: 

`