react-native-camera/react-native-camera #1763: onFacesDetected breaks app on iOS (device-only)

zeekw, 2018-08-27:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Install react-native-camera *with* face detection using only manual procedures (no cocoapods or react-native link)
2. Import { RNCamera, FaceDetector }
3. Add component to page with onFacesDetected callback

### Expected behaviour
It should call the callback method defined

### Actual behaviour
The app completely freezes and loses connection to Xcode (only on hardware. simulator does not crash)

### Environment
- **React Native version**: 0.56.0
- **React Native platform + platform version**: iOS 11.4

### react-native-camera
**Version**: npm version

cgeorgiu, 2018-08-29:
I'm having the same issue but using the cocoapods method.

It does not work on iPhone 5s and iPhone 7 but it works on iPhone 7+
All iPhones are running the same OS version (11.4.1)

cgeorgiu, 2018-08-30:
@possiblytyler Downgraded what to 1.4.0? did you mean downgraded iOS from 11.4.1 to 11.4.0?

sospedra, 2018-10-03:
```
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'FaceDetector'
]
```

Does this works?

sibelius, 2018-10-03:
tks

closed 2018-10-03 by sibelius

imspikey, 2018-10-12:
Same for me using iPhone 6 with iOS 11.4.1

I am using cocapods method but still the app is crashing

tiendn, 2018-10-15:
Same for me using iPad Pro 12.9' with iOS 12.0

imspikey, 2018-10-15:
I have upgraded iPhone to version 12 and the same problem happen ?!!

imspikey, 2018-10-15:
can you please reopen this issue as it looks like many people are having the same issue

tiendn, 2018-10-16:
I have tried switch **cameraType** from `front` to `back`. 
It works fine.

jbwhitcombe, 2018-10-16:
Although for my use case (used to detect if the location of a face in the viewfinder), the front camera needs to work.

tiendn, 2018-10-16:
@jbwhitcombe , me too

imspikey, 2018-10-16:
@tiendn I have tried front and back camera but got the same result EXC_BAD_ACCESS

rajivshah3, 2018-12-03:
This might be related to https://github.com/react-native-community/react-native-camera/issues/1613