react-native-camera/react-native-camera #1765: iOS: Flash flickers when camera starts with flashMode = off

thatjuan, 2018-08-28:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Declare RNCamera with `flashMode = RNCamera.Constants.FlashMode.off`
2. Start the app
3. Flash flickers a couple of times and then turns off

### Expected behaviour
Flash should remain off

### Actual behaviour
Flash flickers a couple of times when the camera loads, then turns off and remains off.

Also, when switching the flashMode to `auto` or `on`, it starts behaving like an always-on torch

### Environment
- **React Native version**: 0.56
- **React Native platform + platform version**: iOS 11.4, iPhone X

### react-native-camera
**Version**: master

I noticed this issue was introduced by https://github.com/react-native-community/react-native-camera/commit/851e6dc57b2bbf5839ed50eabe5e3f557ce20b76 . The commit immediately before it does not have this issue.

ShinsukeWatanabe, 2018-08-28:
> Also, when switching the flashMode to auto or on, it starts behaving like an always-on torch

same issue.

### Environment
- React Native version: 0.55
- Platform: iOS
- react-native-camera: "1.2.0"

fauster01, 2018-08-31:
Same issue.
Flash flickers twice with **any** flashMode (iPhone 8, iPhone 6s).
Also, when switching the flashMode to `auto `or `on`, it starts behaving like an always-on torch on iPhone 6s and like  an always-off torch on iPhone 8.

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: iOS 11.4, iPhone 8, iPhone 6s

### react-native-camera
Version: master

komaldeep, 2018-09-03:
I am also facing the same issue ... flash auto or on behaves like torch in ios

closed 2019-06-08 by stale[bot]