react-native-camera/react-native-camera #1766: [Feature] MSI barcode

zedtux, 2018-08-29:
First of all, thank you guys for this awesome library 👍.

I'm wanted to use it also to scan some bar codes and discovered that actually they are MSI (not code128 as we was expecting).

This issue is a request to implement [the MSI bar codes](https://en.wikipedia.org/wiki/MSI_Barcode) symbology.

phyzalis, 2018-09-07:
Hi guys,

It would really be a great feature

zedtux, 2019-06-01:
OMG ... here we are with those stupid bots ... 🤦‍♂ ... I'm now again in a stupid loop of posting messages in order to keep tickets open 🤕.

I should write a bot to auto post message to block this bot.

zedtux, 2019-07-31:
Uninstall this sucking bot please.

phyzalis, 2019-07-31:
Any news about this dramatically missing feature ?

basselin, 2019-09-29:
Bazinga!

zedtux, 2019-11-28:
Schwarzer Freitag

sibelius, 2019-11-28:
Send pull request

fabriziobertoglio1987, 2020-01-15:
>[MSI (also known as Modified Plessey)](https://en.wikipedia.org/wiki/MSI_Barcode#Character_set_and_binary_lookup) is a barcode symbology developed by the MSI Data Corporation, based on the original Plessey Code symbology. It is a continuous symbology that is not self-checking. MSI is used primarily for inventory control, marking storage containers and shelves in warehouse environments. 

[image]

The [Android Barcode Api](https://developers.google.com/vision/android/barcodes-overview) includes a list of 1 dimensional barcodes:

>1D barcodes: [EAN-13, EAN-8](https://en.wikipedia.org/wiki/International_Article_Number), [UPC-A, UPC-E](https://en.wikipedia.org/wiki/Universal_Product_Code), [Code-39](https://en.wikipedia.org/wiki/Code_39), [Code-93](https://en.wikipedia.org/wiki/Code_93), [Code-128](https://en.wikipedia.org/wiki/Code_128), [ITF](https://en.wikipedia.org/wiki/Interleaved_2_of_5), [Codabar](https://en.wikipedia.org/wiki/Codabar)

`MSI` is not included in the list

zedtux, 2020-03-16:
Find someone to do it.

phyzalis, 2020-05-15:
Still waiting for it :)

zedtux, 2020-09-26:
@stale mark this as a discussion.

basselin, 2021-01-06:
Wait...

phyzalis, 2021-03-08:
Still in discussion no ?

basselin, 2021-05-08:
Hello Stale !

phyzalis, 2021-07-07:
Still wainting

phyzalis, 2021-09-05:
No solution yet

phyzalis, 2021-11-05:
Hasta la vista, baby