react-native-camera/react-native-camera #1767: Duplicate module name: react-native

onmyway133, 2018-08-29:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Follow README and installation guide for iOS, either with automatically with react-native or automatically with pod install, or manually add project and lib

### Actual behaviour
I get error

```
error: bundling failed: Error: jest-haste-map: @providesModule naming collision:
  Duplicate module name: react-native
  Paths: /Users/khoa/XcodeProject2/MyApp/ios/Pods/React/package.json collides with /Users/khoa/XcodeProject2/MyApp/node_modules/react-native/package.json

This error is caused by a @providesModule declaration with the same name across two different files.
    at setModule (/Users/khoa/XcodeProject2/MyApp/node_modules/metro/node_modules/jest-haste-map/build/index.js:446:17)
    at workerReply (/Users/khoa/XcodeProject2/MyApp/node_modules/metro/node_modules/jest-haste-map/build/index.js:496:9)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:188:7)
```

### Environment
react-native-cli: 2.0.1
react-native: 0.56.0

### react-native-camera
"react-native-camera": "^1.2.0"

TheHothihi, 2018-09-12:
The problem occurs when you "pod install" react-native again.
so when nodejs compile the library, nodejs finds that there are two library with same name.
My solution is that open "/Users/khoa/XcodeProject2/MyApp/ios/Pods/React/package.json"
edit "name" field from "react-native" to "react-native2" then run it again.

**And you should edit the package.json in Pods rather than the package.json in node_modules.**

kaueDM, 2018-11-14:
Please update the docs with @n1ru4l answer.
I was unable to build my project using the current `Podfile` example.

n1ru4l, 2018-11-14:
@kaueDM Please feel free to send a pull request!

FullstackJack, 2018-12-05:
Still have the issue after copying @n1ru4l podfile and reinstalling pods.

n1ru4l, 2018-12-05:
@FullstackJack Please provide a minimal reproduction repository.

SteveGreenley, 2019-06-19:
For those that aren't able to resolve this issue by updating their Podfile, you can set up a blacklist in metro.config.js. Here's mine: 

`const blacklist = require('metro-config/src/defaults/blacklist');

module.exports = {
  transformer: {
    getTransformOptions: async () => ({
      transform: {
        experimentalImportSupport: false,
        inlineRequires: false,
      },
    }),
  },
  resolver: {
    blacklistRE: blacklist([
      /node_modules\/.*\/node_modules\/react-native\/.*/,
      /nodejs-assets\/.*/,
      /android\/.*/,
      /\/ios\/.*/
    ])
  },
};`

Please note the last item includes a / before ios. This ensures that libraries like axios aren't included in the blacklist.

tpucci, 2019-08-28:
Try to remove and reinstall your pods, it worked for me:
```
cd ios
rm -rf Pods
pod install
```