react-native-camera/react-native-camera #1768: App (iOS only) crash when camera not authorized in Function as Child Component way

ianc-datacom, 2018-08-30:
### Which implementation are you using
RNCamera

### Steps to reproduce
1. create camera component with FaCC way
2. un-authorized camera in phone setting
3. crash as soon as turning on the camera in the app

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
{({camera, status}) => {
      if (status === RNCamera.Constants.CameraStatus.NOT_AUTHORIZED) {
             return <NoCameraModal />
       } else if (status === RNCamera.Constants.CameraStatus.PENDING_AUTHORIZATION) {
             return <View style={styles.emptyViewFinder} />
       }
       return <Image source={require('~assets/draw/viewfinder.png')} style={styles.viewFinder} />
 }}
Should show the NoCameraModal as expected.

### Actual behaviour
App crash and following error 

-[RNCamera initializeCaptureSessionInput]_block_invoke: Error Domain=AVFoundationErrorDomain Code=-11852 "Cannot use Back Camera" UserInfo={NSLocalizedFailureReason=This app is not authorized to use Back Camera., AVErrorDeviceKey=<AVCaptureFigVideoDevice: 0x109de02c0 [Back Camera][com.apple.avfoundation.avcapturedevice.built-in_video:0]>, NSLocalizedDescription=Cannot use Back Camera}

### Environment
- **React Native v0.55.3**:
-  iOS 11.2.6

### react-native-camera
v1.2.0

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jordanmkoncz, 2018-09-14:
I just ran into this issue as well. Looks like there's some issue using the FaCC (Function as Child Components) approach when the user has not authorized access to the camera.

As a temporary fix, I changed my component to instead use the `pendingAuthorizationView` and `notAuthorizedView` props on `RNCamera`, and then directly included my camera overlay view as a child of `RNCamera`. This seems to achieve the same effect as the FaCC approach, but does not crash when the user has not authorized access to the camera.

Here's a sample of the alternative I'm using as a temporary fix:

```
<RNCamera
  style={styles.cameraContainer}
  type={RNCamera.Constants.Type.back}
  pendingAuthorizationView={
    <View style={styles.pendingAuthorization} />
  }
  notAuthorizedView={
    <View style={styles.notAuthorized} />
  }
>
  <View style={styles.authorized} />
</RNCamera>
```

In case it helps, here's my full raw iOS exception:

```
OS Version: iOS 11.4.1 (15G77)
Report Version: 104

Exception Type: EXC_CRASH (SIGABRT)
Crashed Thread: 18

Thread 18 Crashed:
0   CoreFoundation                  0x3021a7d8c         __exceptionPreprocess
1   libobjc.A.dylib                 0x3006585ec         objc_exception_throw
2   AVFoundation                    0x30d825260         -[AVCaptureSession startRunning]
3   mine                            0x200525048         __24-[RNCamera startSession]_block_invoke
4   libdispatch.dylib               0x3014cfaa0         _dispatch_call_block_and_release
5   libdispatch.dylib               0x3014cfa60         _dispatch_client_callout
6   libdispatch.dylib               0x3014d99b4         _dispatch_queue_serial_drain$VARIANT$mp
7   libdispatch.dylib               0x3014da2fc         _dispatch_queue_invoke$VARIANT$mp
8   libdispatch.dylib               0x3014dacc8         _dispatch_root_queue_drain_deferred_wlh$VARIANT$mp
9   libdispatch.dylib               0x3014e3098         _dispatch_workloop_worker_thread$VARIANT$mp
10  libsystem_pthread.dylib         0x301b36e70         _pthread_wqthread
11  libsystem_pthread.dylib         0x301b36b08         start_wqthread

Thread 0
0   libsystem_kernel.dylib          0x3017b6de8         mach_msg_trap
1   libsystem_kernel.dylib          0x3017b6c60         mach_msg
2   CoreFoundation                  0x30214fe40         __CFRunLoopServiceMachPort
3   CoreFoundation                  0x30214d908         __CFRunLoopRun
4   CoreFoundation                  0x30206dda8         CFRunLoopRunSpecific
5   GraphicsServices                0x306039020         GSEventRunModal
6   UIKit                           0x315d9b758         UIApplicationMain
7   mine                            0x2003c9368         main (main.m:14)
8   libdyld.dylib                   0x30159afc0         start

Thread 1
0   libsystem_kernel.dylib          0x3017d8d78         __workq_kernreturn
1   libsystem_pthread.dylib         0x301b36eb4         _pthread_wqthread

Thread 2
0   libsystem_kernel.dylib          0x3017d8d78         __workq_kernreturn
1   libsystem_pthread.dylib         0x301b370a0         _pthread_wqthread

Thread 3
0   libsystem_kernel.dylib          0x3017d8d78         __workq_kernreturn
1   libsystem_pthread.dylib         0x301b370a0         _pthread_wqthread

Thread 4
0   libsystem_kernel.dylib          0x3017b6e24         semaphore_wait_trap
1   libdispatch.dylib               0x3014d1f48         _dispatch_sema4_wait$VARIANT$mp
[... 124 lines cut ...]
5   libsystem_pthread.dylib         0x301b38110         _pthread_start

Thread 17 name: SentryCrash Exception Handler (Primary)
0   libsystem_kernel.dylib          0x3017b6de8         mach_msg_trap
1   libsystem_kernel.dylib          0x3017b6c60         mach_msg
2   mine                            0x2006bfa80         handleExceptions
3   libsystem_pthread.dylib         0x301b38220         _pthread_body
4   libsystem_pthread.dylib         0x301b38110         _pthread_start

Thread 19
0   libsystem_kernel.dylib          0x3017d8d78         __workq_kernreturn
1   libsystem_pthread.dylib         0x301b36eb4         _pthread_wqthread

Thread 20 name: com.apple.CFSocket.private
0   libsystem_kernel.dylib          0x3017d852c         select$DARWIN_EXTSN
1   CoreFoundation                  0x302158378         __CFSocketManager
2   libsystem_pthread.dylib         0x301b38220         _pthread_body
3   libsystem_pthread.dylib         0x301b38110         _pthread_start

Thread 21
0   libsystem_kernel.dylib          0x3017d8d78         __workq_kernreturn
1   libsystem_pthread.dylib         0x301b36eb4         _pthread_wqthread

EOF
```

ianc-datacom, 2018-09-17:
Thanks for your comment @jordanmkoncz, I have done the same in my work two days after this issue got ignored. I guess this isn't important enough to get attention.

Zabi-Progton, 2019-02-27:
I have also run into this issue, this issue needs some serious attention.

jmayalag, 2019-04-10:
There should be a notice in the docs to avoid using FaCC

closed 2019-06-16 by stale[bot]

cristianocca, 2019-07-19:
Facing the same issue, why close this? ? FaCC is very useful and it is a silly crash.

skoob13, 2019-08-08:
I have the same issue using FaCC