react-native-camera/react-native-camera #1769: Unable to record video on android

jonathangreco, 2018-08-30:
### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. Using this component

```
import React from "react";
import { RNCamera } from "react-native-camera";
import { StyleSheet, TouchableOpacity, View} from "react-native";
import {Icon} from "native-base";

/**
 * This is the component wich allow the user to film his own film
 */
export default class CameraScreen extends React.Component
{
  constructor(props) {
    super(props);
    this.camera = null;
  }

  state = {
    isRecording: false,
    iconVid: "play",
    iconMode: null,
    overlay: true
  };

  renderCamera = () => {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={cam => {
            this.camera = cam;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.off}
          permissionDialogTitle={"Permission to use camera"}
          permissionDialogMessage={
            "We need your permission to use your camera phone"
          }
        >
          <View>
            <TouchableOpacity onPress={() => this.recordVideo()}>
              <Icon name={this.state.iconVid} type={"FontAwesome"} style={styles.iconCamera}/>
            </TouchableOpacity>
          </View>
        </RNCamera>
      </View>
    );
  };

  recordVideo = () => {
    if (this.camera) {
      if (this.state.isRecording === false) {
        this.setState({ isRecording: true, iconVid: "stop" });
        this.camera.recordAsync().then(data => {
          console.log(data);
        }).catch((err) => {
          console.log(err);
        });
        console.log("Device is Recording");
      } else {
        this.camera.stopRecording();
        this.setState({
          isRecording: false,
          iconVid: "play"
        });
        console.log("Stop Recording");
      }
    }
  };

  render() {
    return this.renderCamera()
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    marginTop:30,
  },
  preview: {
    flex: 1,
    justifyContent: "center",
    flexDirection: "row",
    alignItems: "flex-end",
  },
  iconCamera: {
    color: "#ff4459",
    marginBottom: 50,
  }
});

```

### Expected behaviour
The camera should record something.

### Actual behaviour

See the following screen to see what happenened (console log are explicit)
[image]

### Environment

    "react-native": "0.56.0",
    "react-native-camera": "^1.2.0",
    Android 7, API 24 (texted on Samsung SM-T580

### react-native-camera

    "react-native-camera": "^1.2.0",

jonathangreco, 2018-08-30:
Additionnal informations:

Tested on Ipad with RNCamera -> crashes at preview.
Tested on Ipad with RCTCamera -> it worked.
Tested on Android with RCTCamera -> same issue.

sibelius, 2018-08-30:
you should use RNCamera

jonathangreco, 2018-09-19:
This can be close as the issue was not relevant with RNCamera.

closed 2018-09-19 by jonathangreco