react-native-camera/react-native-camera #1774: android vivoY67 can't read code-39 or code-128

sybil052, 2018-09-03:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using
<RNCamera
                                ref={(cam) => {
                                    this.camera = cam;
                                }}
                                style={styles.cameraStyle}
                                barcodeScannerEnabled={true}
                                onBarCodeRead={
                                    this.barcodeReceived
                                }
                                flashMode={openFlash ? RNCamera.Constants.FlashMode.on : RNCamera.Constants.FlashMode.off}
                                // torchMode={openFlash ? 'on' : 'off'}
                            >

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?

### Expected behaviour
 function onBarCodeRead can return barcode.

### Actual behaviour
 when vivoY67 read code-39 or code-128, function onBarCodeRead doesn't work.

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: vivo Y67 Android 6.0

### react-native-camera
**Version**: "react-native-camera": "git+https://github.com/react-native-community/react-native-camera.git"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

nadzic, 2018-09-16:
Did you get black screen or simply did not read the code?

sibelius, 2019-04-18:
try latest version

it uses MLKit

closed 2019-06-24 by stale[bot]

abhi-nahi, 2019-10-24:
Do I need to integrate the firebase ML toolkit to use this feature ???