react-native-camera/react-native-camera #1778: Can't record short videos on iOS. Crahses.

diego-caceres, 2018-09-04:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

RNCamera

### Steps to reproduce
1. Call recordAsync on the camera reference
2. Wait 1 second and call stopRecording on camera reference
3. The promise from recordAsync throws an exception

### Does it work with Expo Camera?
I haven't tried

### Expected behavior
The short video should be recorded successfully 

### Actual behavior
Throws an exception

### Environment
- **React Native version**: 0.53.2
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: master

On Android it works fine.

n1ru4l, 2018-09-24:
@diego-caceres Could you please provide crash logs?

watadarkstar, 2018-10-05:
I also get this issue and can provide logs.

n1ru4l, 2018-10-09:
@watadarkstar Please provide crash logs

watadarkstar, 2018-10-09:
Hi @n1ru4l you can find the logs here: https://github.com/react-native-community/react-native-camera/issues/1823

n1ru4l, 2018-10-10:
Duplicate of #1823

closed 2018-10-10 by n1ru4l