react-native-camera/react-native-camera #1785: Async error java takePictureAsync()

rusliabdulgani, 2018-09-10:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

RNCamera

### Steps to reproduce
1. Open Camera
2. camera.takePictureAsync()
3. force closed (sometimes OK)

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
After execute promise takePictureAsync() it should be take picture

### Actual behaviour
After execute promise/await takePictureAsync() app force closed.
if i used crashlytics it found error like this.

`Fatal Exception: java.lang.RuntimeException: An error occurred while executing doInBackground()
       at android.os.AsyncTask$3.done(AsyncTask.java:325)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
       at java.util.concurrent.FutureTask.run(FutureTask.java:242)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
       at java.lang.Thread.run(Thread.java:761)`

### Environment
- **React Native version**: 0.55.4
- **React Native platform + platform version**: Android version 7.1.1 

### react-native-camera
**Version**: "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

rusliabdulgani, 2018-09-17:
hi all, 

anybody know this issue?

SimonErm, 2018-09-25:
do you use fixOrientation option?

n1ru4l, 2018-12-26:
Closing due to missing feedback.

closed 2018-12-26 by n1ru4l