react-native-camera/react-native-camera #1787: Unable

telkandrew, 2018-09-11:
After installing react-native-camera via "npm install react-native-camera", and then running react-native-link, my build is logging the following errors in Android Studio.

Unable to resolve dependency for ':react-native-camera@debug/compileClasspath': Could not resolve com.facebook.infer.annotation:infer-annotation:+.  

and...

Unable to resolve dependency for ':react-native-camera@release/compileClasspath': Could not resolve com.facebook.infer.annotation:infer-annotation:+.

Running React-Native-Camera v1.2.0

jgcmarins, 2018-09-11:
try to link only this package, i.e.: `react-native-link react-native-camera`

fadlykayo, 2019-04-23:
Having the same problem. Any solution for this?