react-native-camera/react-native-camera #1791: CocoaPods install failed

yli18, 2018-09-13:
Text Recognition is a great function, I tried to integrate it with this version
github:react-native-community/react-native-camera#e12310869ebefb1a264abb593126a52481220c7e

I used to automatic install with react-native which has no problem. However, CocoaPods Path is the only option for Text Recognition. So I follow the document to Install with CocoaPod.  

### Steps to reproduce
1. npm install    
"react-native-camera": "github:react-native-community/react-native-camera#e12310869ebefb1a264abb593126a52481220c7e"
2. edit pod file 
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'TextDetector'
]
3. pod install  [!] React has been deprecated
4 compile

### Expected behaviour
successfully compiled 

### Actual behaviour

1. bundle console keeps warning while compiling
This warning is caused by a @providesModule declaration with the same name across two different files.
jest-haste-map: @providesModule naming collision:
2. build failed 
'React/RCTBridge.h' file not found 

I also tried to edit the package.json file("name": "react-native"   to  "name": "react-native2") in Pods,  but it doesn't help

### Environment
- **React Native version**: 0.54
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: e12310869ebefb1a264abb593126a52481220c7e

yli18, 2018-09-13:
Update, I add the following code to Podfile
    pod 'React', :path => '../node_modules/react-native',:subspecs => [
    'DevSupport',
    'RCTNetwork',
    ]
    pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

and set Dead Code Stripping to yes, build success with run time error 
error: bundling failed: ambiguous resolution: module `.../index.js` tries to require `react-native`, but there are several files providing this module. You can delete or fix them: 

It is still not solved. RN with Pod is often tricky

yli18, 2018-09-14:
Anyone succeed in installing the library with Coco pls leave comments, how you config the Podfile

I spend hours still stuck here.

dov11, 2018-09-14:
Try with the following in your podfile:
```
target 'yourTargetName' do

  pod 'React', :path => '../node_modules/react-native', :subspecs => [
    'Core',
    'CxxBridge', # Include this for RN >= 0.47
    'DevSupport', # Include this to enable In-App Devmenu if RN >= 0.43
    'RCTText',
    'RCTNetwork',
    'RCTWebSocket', # Needed for debugging
    'RCTAnimation', # Needed for FlatList and animations running on native UI thread
    # Add any other subspecs you want to use in your project
  ]

  # Explicitly include Yoga if you are using RN >= 0.42.0
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

  # Third party deps podspec link

  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'TextDetector'
  ]

end

post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == "React"
      target.remove_from_project
    end
  end
end
```

yli18, 2018-09-14:
@dov11  It works like a charm! Text recognition works!!
Thank you, you saved my day!

closed 2018-09-14 by yli18

raygerrard, 2018-09-18:
textRecogniser still doesn't work for me using the above:

I get the following error doing pod install:

```sh
[!] CocoaPods could not find compatible versions for pod "react-native-camera/TextDetector":
  In Podfile:
    react-native-camera/TextDetector (from `../node_modules/react-native-camera`)

None of your spec sources contain a spec satisfying the dependency: `react-native-camera/TextDetector (from `../node_modules/react-native-camera`)`.
```

dov11, 2018-09-18:
Hi @raygerrard , are you using react-native-camera from master? Could you please check if the file `node_modules/react-native-camera/react-native-camera.podspec` in your project contains the following line: `s.subspec "TextDetector" do |ss|` (line 38)

shreyasb07, 2019-04-02:
@dov11  I am getting the same error as @raygerrard and yes I am using react-native-camera from master. I also have the s.subspec "TextDetector" do |ss| line in the .podspec file.

I'd really appreciate some help on this, as I'm a newbie in the world of RN, and don't have much idea about using Cocoapods.
Thanks!

dov11, 2019-04-05:
Hi @shreyasb07 have you tried building [mlkit example](https://github.com/react-native-community/react-native-camera/tree/master/examples/mlkit)? (you would need to set-up a firebase project following the steps [here](https://github.com/react-native-community/react-native-camera#additional-steps-for-text-recognition), as we have migrated to Firebase MLKit).

If it works, I would suggest trying to use the latest version of react-native-camera in your own project, removing Pods (rm -rf ios/Pods) and following installation steps.

If it does not help, please post your Podfile and info about your RN/Xcode versions.