react-native-camera/react-native-camera #1797: Black screen after first scanning on Android 7.0, 8.1

nadzic, 2018-09-16:
### Warning

### Which implementation are you using

*RNCamera* 

### Steps to reproduce
1. Coming to scanning screen (I am using react-navigation)
2. Scan product bacode
3. After scanning black screen

### Does it work with Expo Camera?
We did not try yet expo camera.
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
On iOS it works normally, after scanning I can scan normally next product. 

### Actual behaviour
On Android after I scan one product it show black screen and I cannot scan next product. If I go to next screen and return back (to reset navigation state) then I can scan next product again, but after first scanning.

### Environment
- **React Native version**: 0.55
- **React Native platform + platform version**: Android 7.0, Android 8.1

### react-native-camera
**Version**: 1.2.0 (tried also 1.1.2)

### example of our component

`    <SideMenu
        menu={menu}
        isOpen={isOpen}
        onChange={isOpen => this.updateMenuState(isOpen)}
      >
        <View style={styles.container}>
          <RNCamera
            ref={ref => {
              this.camera = ref;
            }}
            style={styles.cameraPreview}
            onBarCodeRead={(!currentKitchenVisible && rnCameraEnable) ? this.onBarCodeRead : null}
            permissionDialogTitle={'Permission to use camera'}
            permissionDialogMessage={'We need your permission to use your camera phone'}
          >
            <View style={styles.logoWrapper}>
              <Image
                style={styles.logo}
                source={require('./img/logo.png')}
              />
            </View>
            {isEmulator &&
              <View style={styles.testWrapper}>
                <TextInput
                  style={{ width: 300, backgroundColor: COLORS.LIGHTGREY, height: 35 }}
                  placeholder="Upc kitchen number"
                  underlineColorAndroid="transparent"
                  autoCapitalize="none"
                  placeholderTextColor={COLORS.LIGHTGREY}
                  value={testUpcCode}
                  onChangeText={(text) => this.setState({ testUpcCode: text })}
                />
              </View>
            }
            {!isEmulator && <OverlayMask {...{ loading }} />}
            {isEmulator && !currentKitchenVisible &&
              <View style={[styles.testWrapper, { top: 250 }]}>
                <TouchableOpacity
                  style={styles.testButton}
                  onPress={() => this.onBarCodeRead()}
                >
                  <Text style={styles.testText}>Apply</Text>
                </TouchableOpacity>
              </View>
            }
            <TouchableOpacity
              onPress={this.toggle}
              style={styles.menuButton}
            >
              <Image
                source={require('./img/menu.png')}
                style={styles.menu}
              />
            </TouchableOpacity>
            <View style={styles.scanTextWrapper}>
              <Text style={styles.scanText}>Scan kitchen barcode</Text>
            </View>
            {currentKitchenVisible &&
            <View style={styles.kitchen}>
              <Text style={styles.kitchenText}>{companyName}</Text>
              <View style={{ marginLeft: 15 }}>
                <Image
                  source={require('./img/check.png')}
                  style={styles.check}
                />
              </View>
            </View>
            }
            {(!isNil(currentKitchen.company_id) && !isNil(currentKitchen.company_name)) &&
            <TouchableOpacity
              style={styles.scanProductsButton}
              onPress={() => {
                this.props.navigation.dispatch(resetActionGoToScanProduct);
                }
              }
            >
              <Text style={styles.scanProductsText}>{`Scan products`}</Text>
            </TouchableOpacity>
            }
          </RNCamera>
        </View>
      </SideMenu>`

- We already tried to remove SideMenu and other View that RNCamera was top wrapper, did not help,

- We also tried without variables (disable/enable camera): `onBarCodeRead={(!currentKitchenVisible && rnCameraEnable) ? this.onBarCodeRead : null}`,

- We also tried downgrade camera, upgrade react-native from 0.52 to 0.55, did not help.

What else should we try?

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-09-25:
Can you tell me what your this.onBarCodeRead does? Should pop up a modal or do you just save scanned data?

GaneshSaiKumar, 2019-01-10:
Duplicate of #1686

closed 2019-06-08 by stale[bot]