react-native-camera/react-native-camera #1805: {"framesToPop":1,"code":"E_CAPTURE_FAILED","line":21762,"column":31}

meiqi1992, 2018-09-20:
### Warning
RCTCamera is **DEPRECATED** on v1.0.0 follow our [Migration guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

### Which implementation are you using

*RNCamera* or RCTCamera (RCTCamera will be removed on v2.0.0)?

### Steps to reproduce
1.
2.
3.

### Does it work with Expo Camera?
Check usage with Expo - https://github.com/react-native-community/react-native-camera/blob/master/docs/Expo_Usage.md
You should open an issue there as well, so we can cooperate in a solution.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

android 7.0
react-native 0.57
react-native-camera 1.2.0
react-navigation 2.14.0

code:
 this.camera.recordAsync().then((data) => {
    
        }).catch(err => {
            console.warn('getCatch')
            console.warn(err)
        });

then it get catch

meiqi1992, 2018-09-20:
need help

sibelius, 2018-09-20:
check this https://medium.com/react-native-training/uploading-videos-from-react-native-c79f520b9ae1

and this

https://medium.com/react-native-training/build-youtube-alike-livestreams-with-react-native-8dde24adf543

io-pan, 2018-12-06:
I had the same error and solved it after reading [documentation](https://github.com/react-native-community/react-native-camera#user-content-permissions):    ;-)

Add <uses-permission android:name="android.permission.RECORD_AUDIO"/>
in AndroidManifest.xml

n1ru4l, 2018-12-11:
Like @io-pan explained you have to add `<uses-permission android:name="android.permission.RECORD_AUDIO"/>` to your `AndroidManifest.xml`.

Your adb logcat logs should show something like this:
```
2018-12-11 11:12:05.418 3712-3987/? E/MediaRecorderService: Request requires android.permission.RECORD_AUDIO
```

closed 2019-04-18 by sibelius