react-native-camera/react-native-camera #1807: Android: captured image is always rotated

BigPun86, 2018-09-20:
### Which implementation are you using

*RNCamera*

### Steps to reproduce
1. Take a picture with android
    const data = await this.camera.takePictureAsync(this.options); 
    if(data && data.uri) this.setState({ image: data.uri } )
2. display the image within a Image component
3. Check the rotation of the image => in my case it is always rotated by 90deg

### Expected behaviour
The image should not be rotated

### Actual behaviour
The image is rotated by 90deg

### Environment
- **React Native version 53.3**:
- **Google Nexus 5X - API 26 | Samsung Galaxy S5 | API 19

### react-native-camera
**Version**: 1.2.0

XaviSmith, 2018-10-11:
+1, also getting this error

mcorner, 2018-12-14:
Even with 1.6.3 (which contained a fix), this is still broken for me on some android devices (Samsung for instance).

const options = { quality: 0.5, base64: true, orientation: RNCamera.Constants.ORIENTATION_UP };
const data = await this.camera.takePictureAsync(options);

fixOrientation: true does work, but it is incredibly slow (6 seconds) because it is rotating the image afterwards.

scarlac, 2019-03-13:
There seems to be general issues with orientation on both iOS and Android. It seems we are adding more and more properties to workaround these issue (`orientation`, `fixOrientation`, `forceUpOrientation`) while the underlying issue remains. The default setup is broken so it would be really great if these issues were fixed, as it's not viable fix to add additional processing to taking photos (it's already quite slow).

sibelius, 2019-05-09:
can you try latest version

we fixed some orientation issues again

andresmtz98, 2019-05-09:
https://stackoverflow.com/questions/14066038/why-does-an-image-captured-using-camera-intent-gets-rotated-on-some-devices-on-a

This is a global issue

asanz8, 2019-05-22:
I fixed this behavior on Samsung S3 Mini using:

`orientation: 'portrait', fixOrientation: true`

Version 2.9.0

guildeassis, 2019-06-15:
I solved this problem, but it was not so simple. You need to use: `exif: true`  and get the `exif.Orientation` like it:
```
takePicture = async() => {
    ...
      const options = { exif: true };
      const data = await this.camera.takePictureAsync(options);
      this.setState({orientValue: data.exif.Orientation});
    ...
}
```
Now, you can use `this.state.orientValue` inside a function and customize the Image like this:
```
<Image
    style={{ transform: [{ rotate: this.rotateImage() }] }}
/>
```
And finally, rotateImage:
```
rotateImage() {
    let exifOrientation = this.state.orientValue;
    let degRotation;
    switch (exifOrientation) {
	case 3:
	       degRotation = "180deg"
	       break;
	case 4:
		degRotation = "180deg"
		break;
	case 5:
		degRotation = "90deg"
		break;
	case 6:
		degRotation = "90deg"
		break;
	case 7:
		degRotation = "270deg"
		break;
	case 8:
		degRotation = "270deg"
		break;
	default: 
		degRotation = "0deg"
	}
   return degRotation;
}
```
Why rotate this deg? The image explain:
[image]

Note: this only rotates and does not reverse. I think it will work on most cell phones. Am I right?
Note 2: If you save the image to a server, then you need to rotate there as well. That way it only rotates in the react native app. In my case, i use laravel intervention for fix the orientation in the back-end.

Michaelvilleneuve, 2019-07-03:
Hi, still seeing this issue on a Samsung Galaxy Note 3, Android 5.0
It appears that it is very specific to some Android and particularly samsung devices as always. 
Here is how we are taking the photo : 
```
const data = await this.camera.takePictureAsync({
                quality: 0.4,
                base64: true,
                width: 1080,
                fixOrientation: true,
});
```

MateusAndrade, 2019-07-15:
@Michaelvilleneuve you tested if ```writeExif``` solves this? I remember having a similar problem, and using this with ```fixOrientation``` worked for me.

fabriziobertoglio1987, 2019-07-16:
> https://stackoverflow.com/questions/14066038/why-does-an-image-captured-using-camera-intent-gets-rotated-on-some-devices-on-a
> 
> This is a global issue

in fact [version 2.9.0 of ReactNativeCamera adds those Exif feature](https://github.com/react-native-community/react-native-camera/releases/tag/v2.9.0) from the above stackoverflow discussion

closed 2019-08-02 by sibelius

n1ru4l, 2019-08-02:
:tada: This issue has been resolved in version 3.0.1 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.0.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

HappyCMD, 2019-10-02:
@n1ru4l I dont think it has, I'm having the same issue on v 3.6.0

HauLuu, 2019-10-15:
@n1ru4l It actually not, I'm on v3.7.2

      const options = {
        quality: ,
        base64: true,
        orientation: RNCamera.Constants.Orientation.auto,
        pauseAfterCapture: true,
        fixOrientation: true
      };
      const data = await this.camera.takePictureAsync(options);

khaledalam, 2021-02-15:
this will save your time :)
```
const options = {
          quality: 1,
          base64: true,
          orientation: RNCamera.Constants.Orientation.portrait,
          fixOrientation: true
};
const data = await props.mycamera.current.takePictureAsync(options);
```

frankcodev, 2021-08-14:
Hi, this worked for me:

`const options = { quality: 0.5, base64: true,  mirrorImage: true, fixOrientation: true };`

vipu-sa, 2021-08-19:
> I fixed this behavior on Samsung S3 Mini using:
> 
> `orientation: 'portrait', fixOrientation: true`
> 
> Version 2.9.0

I have not working this in android.

ashishprasad2163, 2021-09-21:
I was getting this error on using the front camera only. 
so this worked for me,
        mirrorImage: frontCamera  ? true : false,
        forceUpOrientation: frontCamera  ? true : false,
        fixOrientation: frontCamera  ? true : false,

rajan-keypress, 2022-03-04:
const data = await takePicture({
        mirrorImage: true,
      }); work for me