react-native-camera/react-native-camera #1813: IOS scanning bar code is very difficult

qingjiaowochengzd, 2018-09-22:
### Which implementation are you using

*RNCamera*
"react": "16.3.1",
"react-native": "0.55.4",
"react-native-camera": "1.2.0",

my code:
import Camera from 'react-native-camera';

barcodeReceived = (e) =>{}

<Camera
                    ref={(cam) => {
                        this.camera = cam;
                    }}
                    style={styles.preview}
                    aspect={this.state.camera.aspect}
                    onBarCodeRead={this.barcodeReceived.bind(this)}
                >
                <View style = {{height: Platform.OS == 'ios' ? (height-264)/3:(height-244)/3,width:width,backgroundColor:'rgba(0,0,0,0.5)',}}>
                </View>
                    <View style={{flexDirection:'row'}}>
                        <View style={styles.itemStyle}/>
                            <ImageBackground style={styles.rectangle}
                                >
                                <Animated.View style={[styles.animatiedStyle, {
                                    transform: [{
                                        translateY: this.state.anim.interpolate({
                                            inputRange: [0,1],
                                            outputRange: [0,200]
                                        })
                                    }]
                                }]}>
                                </Animated.View>
                            </ImageBackground>
                        <View style={styles.itemStyle}/>
                    </View>
                    <View style={{flex:1,backgroundColor:'rgba(0, 0, 0, 0.5)',width:width,alignItems:'center'}}>
                        <Text style={styles.textStyle}>将二维码放入框内,即可自动扫描</Text>
                    </View>
                </Camera>

### Expected behaviour
Running Android and iOS locally can sweep the qrcode, but they sweep out the barcode Almost impossible. After packing out, you can sweep out the qrcode, Android can sweep out the barcode, but IOS can not sweep out the barcode.

### Environment
ios 10.2.1
android 7.0

qingjiaowochengzd, 2018-09-22:
Camera not recognising barcodes on iOS until you rotate the phone #1706

n1ru4l, 2018-09-24:
duplicate of #1706

closed 2018-09-24 by n1ru4l

yorickshan, 2021-12-30:
I'm also having this issue