react-native-camera/react-native-camera #1821: recordAsync doesn't resolve after 60 seconds

piuccio, 2018-09-26:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**
- iOS: 
- react-native-camera: 1.2.0
- react-native: 0.57.1
- react: 16.5.0

**Description/Current Behaviour**

I'm using the code below. It's supposed to stop recording after some time (`setTimeout`).
Works fine for values below 1 minute, but as soon as I record for longer than 60 seconds the promise from `recordAsync` never resolves.

```js
class Recorder extends React.PureComponent {
  startRecording = () => {
    console.log('starting');
    this.camera
      .recordAsync({
        quality: RNCamera.Constants.VideoQuality['480p'],
        mirrorVideo: true,
      })
      .then(this.onCaptureEnd)
      .catch((err) => console.log('err', err));
    setTimeout(this.stopRecording, 70 * 1000);
  }

  stopRecording = () => {
    console.log('stopping');
    this.camera.stopRecording();
  }

  onCaptureEnd = (data: RecordResponse) => {
    console.log('got data');
    this.props.onRecordComplete(data.uri);
  }

  render() {
    return (
      <RNCamera
        captureAudio
        permissionDialogMessage={this.props.intl.formatMessage(commonMessages.androidPermissionMessage)}
        permissionDialogTitle={this.props.intl.formatMessage(commonMessages.androidPermissionTitle)}
        ref={(cam) => { this.camera = cam; }}
        style={styles.backgroundVideo}
        type={RNCamera.Constants.Type.front}
        videoStabilizationMode={RNCamera.Constants.VideoStabilization.auto}
        onCameraReady={this.startRecording}
        onMountError={(err) => console.log('mount error', err)}
      />
    );
  }
}
```

When I inspect the logs I can see that `calling` and `stopping` are logged, but `got data` never happens. Nothing else is logged, no errors seems to be thrown.

**Expected Behaviour**
Expecting to record any length 

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-09-26:
Have you tried using maxDuration record option instead of using a timeout?

piuccio, 2018-09-26:
Well, the setTimeout is there just to automate the example, if I click on the stop button after 1 minute I'm still stuck in a never ending recording

SimonErm, 2018-09-26:
Can you try without mirroVideo option?

BaderSerhan, 2018-10-22:
@piuccio yes it mirrors the output video. It was also the problem to my issue :/
#1848

albinekb, 2020-03-26:
I can confirm this is still an issue on the latest version

albinekb, 2020-06-02:
This is still an issue with `mirrorVideo: true`

neysi, 2020-10-05:
Promise is not resolved when mirror video is true.

closed 2021-01-13 by stale[bot]

sheyko-d, 2021-03-17:
Any updates here? It's still not resolving after 60 seconds, if the **mirrorVideo** is set to **true**

pragadeeshk, 2021-06-27:
I was able to solve this by moving the resolve block in captureOutput method in RNCamera.m

```
if ([connections[0] isVideoMirrored]) {
    [self mirrorVideo:outputFileURL completion:^(NSURL *mirroredURL) {
        result[@"uri"] = mirroredURL.absoluteString;
    }];
    resolveBlock();
    return;
}
```

swushi, 2021-07-19:
`mirrorVideo` was also the issue for me, this is still an issue on `3.44.3`

MrXyfir, 2021-11-15:
#3312