react-native-camera/react-native-camera #1822: iOS: build error when follow the 'No Face Detection steps'

Jeepeng, 2018-09-27:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

**Platforms**
iOS

**Versions**

- Android:
- iOS:
- react-native-camera: v1.3.0
- react-native:
- react:

**Description/Current Behaviour**
error log:
```
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$_RNSensorOrientationChecker", referenced from:
      objc-class-ref in libRNCamera.a(RNCamera.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
**cause: `/postinstall_project/projectWithoutFaceDetection.pbxproj` is outdated**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

SimonErm, 2018-09-27:
https://github.com/react-native-community/react-native-camera/pull/1762

sibelius, 2018-09-28:
can somebody update `/postinstall_project/projectWithoutFaceDetection.pbxproj`?

Willis0826, 2018-10-01:
I saw the latest version @1.3.0 has been merged, and I tried to update & build.
Unfortunately, I still got the same error : 
```
Undefined symbols for architecture x86_64:
    "_OBJC_CLASS_$_RNSensorOrientationChecker", referenced from:
        objc-class-ref in libreact-native-camera.a(RNCamera.o)
```

```
react-native-camera: 1.3.0
react-native: 0.57.1
```

sibelius, 2018-10-01:
try with master version

Willis0826, 2018-10-01:
@sibelius I tried the master version : 
`react-native-camera: git://github.com/react-native-community/react-native-camera.git#master`

muratozgul, 2018-10-03:
same problem here (after upgrading to Xcode 10)

Willis0826, 2018-10-03:
@muratozgul I tried the steps you mentioned, but couldn't fix it.

sibelius, 2018-10-03:
tks

closed 2018-10-03 by sibelius

youssefatti, 2018-10-11:
none of those things worked for me. Any update?

sospedra, 2018-10-11:
@youssefatti please, consider opening a new issue including the details of your problem 😀