react-native-camera/react-native-camera #1826: Android: Error while updating property 'pictureSize'

toeyanuntachai, 2018-10-01:
# Bug Report

**Versions**

"react-native-camera": "^1.3.0",

**Description/Current Behaviour**

I found this error when I'm navigated to the screen that makes for using the RNCamera

[image]
Here is my code
[image]

toeyanuntachai, 2018-11-18:
@Ratnesh-Kumar Yes, I have resolved this issue, because I have installed some package that using the same dependency "com.google.android.cameraview", so looks like it's will confuse to resolve the correct package.

P.S. Sorry for English, I'm trying to practice it.

closed 2018-11-18 by toeyanuntachai

Ratnesh-Kumar, 2018-11-20:
I am trying firebase with RNCamera getting this error, trying to resolve.

n1ru4l, 2018-11-20:
@Ratnesh-Kumar We cannot help you with clashing dependencies. It seems like you  have to force a specific version for all resolved `com.google.android.cameraview` dependencies in your `build.gradle`.

sibelius, 2018-11-20:
we need more docs about this topic

n1ru4l, 2018-11-20:
@Ratnesh-Kumar Can you please provide a reproduction repository so I can figure out a solution that can be documented?