react-native-camera/react-native-camera #1830: How to natively draw over the camera view?

Tsiniloiv, 2018-10-02:
# Question

Heya! So, I've gotten tap-to-focus to mostly work on the Android side of RNCamera, and now I'm looking into making it a bit more apparent by drawing a circle where the user touches the camera view.

Overlaying a view over the Camera component doesn't seem to be the right answer, since that blocks the touch from reaching the actual Camera component. So it looks like I need to do this natively.

Unfortunately I'm not super familiar with doing stuff natively in Android. So far I've been trying to override the CameraView onDraw function, but that only seems to work once, until the preview loads and seems to draw over everything else. Tried throwing invalidate() into onDraw to get like a constant thing going, and that seemed to work for a second, but then it crashed, so I'm pretty sure that's not the right way.

Ideally I'd like to just tie it to the onTouch function, but I don't seem to have access to the Canvas from there.

I've also been looking into TextureViewPreview, which seems to have a callback for every new frame, but I'm not quite sure what to do with that either.

I did find [this thing](https://stackoverflow.com/questions/31173476/android-sdk-camera2-draw-rectangle-over-textureview), which seems very close to what I want, but I'm not quite seeing how to translate it into something that would work with how RNCamera is structured.

How do you natively draw over the camera preview for Android?

**Tags**
#android #taptofocus #preview #java

jgfidelis, 2018-10-04:
I don't  know exactly how to do it on android either.

On iOS, I guess it would be not too dificult to draw a UIView over the preview layer. There was a PR once upon a time that implemented but nobody ever tested and we did not merge. You can try to look for it.

msageryd, 2018-10-25:
Why cant you just draw a regular view with borders (i.e. a rectangle). Use absolute position and pointerEvents='none'. This should not block the touches from reaching the camera view.

Something like this (not tested)
```
<View 
  pointerEvents={'none'} 
  style={{
    borderWidth: 1, 
    borderColor: 'yellow', 
    position: 'absolute', 
    left: 100, 
    top: 100, 
    width: 20, 
    height: 20
  }}
>
```

Craigtut, 2018-11-21:
@Tsiniloiv - How did you get tap-to-focus on Android to work? Only iOS appears to be supported. Did you build it in yourself?

Tsiniloiv, 2018-11-21:
@Craigtut Built it in myself based on some Stack Overflow answers. How to do it:

Drop this into com.google.android.cameraview.CameraView

```
@Override
    public boolean onTouchEvent(final MotionEvent event) {
      if (event.getAction() == MotionEvent.ACTION_DOWN) {
        float x = event.getX();
        float y = event.getY();
        float touchMajor = event.getTouchMajor();
        float touchMinor = event.getTouchMinor();

        Rect touchRect = new Rect((int)(x - touchMajor / 2), (int)(y - touchMinor / 2), (int)(x + touchMajor / 2), (int)(y + touchMinor / 2));

        Rect focusAreaRect = new Rect();

        focusAreaRect.set(touchRect.left * 2000 / this.getView().getWidth() - 1000,
          touchRect.top * 2000 / this.getView().getHeight() - 1000,
          touchRect.right * 2000 / this.getView().getWidth() - 1000,
          touchRect.bottom * 2000 / this.getView().getHeight() - 1000);

        Camera.Area focusArea = new Camera.Area(focusAreaRect, 1000);

        mImpl.submitFocusAreaRect(focusArea);
        return true;
      }
      return false;
    }
```

Drop this into com.google.android.cameraview.Camera1:

```
public void submitFocusAreaRect(final Camera.Area focusArea)
  {
    if (mCameraParameters.getMaxNumFocusAreas() == 0) { return; }
    // Submit focus area to camera
    ArrayList<Camera.Area> focusAreas = new ArrayList<Camera.Area>();
    focusAreas.add(focusArea);

    mCameraParameters.setFocusMode(Camera.Parameters.FOCUS_MODE_AUTO);
    mCameraParameters.setFocusAreas(focusAreas);

    mCamera.setParameters(mCameraParameters);

    // Start the autofocus operation
    mCamera.autoFocus(new Camera.AutoFocusCallback() {
      @Override
      public void onAutoFocus(boolean success, Camera camera) {
        mCamera.cancelAutoFocus();
      }
    });
  }
```

Add an abstract submitFocusAreaRect to the CameraViewImpl class, and stub it in Camera2 and Camera2Api23.

pedrosimao, 2019-01-07:
Nice, so, it basically means you can autofocus with your thinger on Android (and iOs) but you want a visual element (square or round) that is drawn when user touches the screen?
If it is the case it won't be difficult to create a prop that provides a visual overlay for helping with focus. I can try to make one...

closed 2019-06-08 by stale[bot]